linux-can/can-utils #158: Networking with CAN-J1939 support (Manjaro Linux)

josemic, 2019-10-24:
I am using kernel linux54rc2.d1006.gda0c9 (experimental).

As described in can-j1939-kickstart.md the kernel module is loaded as following:
$ sudo modprobe vcan
$ sudo ip link add can0 type vcan
$ sudo ip link set can0 up
$ sudo modprobe can-j1939
modprobe: FATAL: Module can-j1939 not found in directory /lib/modules/5.4.0-1-MANJARO

./testj1939 -r can0:
./testj1939: socket(j1939): Protocol not supported

It is unclear to me, whether this has to do with the kernel package from majaro why the module is not there and is not compiled or where to find the kernel module can-j1939.ko and how to compile it.

I am cross positing it here:
https://forum.manjaro.org/t/networking-with-can-j1939-support/108403

hartkopp, 2019-10-24:
Are you sure the CAN_J1939 support is enabled in your 5.4-rc2 kernel?
There is no "default y" tag in the linux/net/can/j1939/Kconfig - so maybe it is simply not selected to build.

hartkopp, 2019-10-25:
As @josemic opened a duplicated issue here https://github.com/linux-can/can-utils/issues/159 and the Debian kernel build obviously has CAN_J1939 support enabled, we can close this Manjaro specific issue.

closed 2019-10-25 by hartkopp