linux-can/can-utils #160: Time delay in receiving CAN message

Prj1508, 2019-10-25:
I have socketcan running at 500 baud and I am experiencing delays/lag in receiving the CAN messages. I am testing onboard a vehicle and when I turn the steering wheel for ex, there is 5-10 seconds delay in updating the data field of the respective message in socketcan(in candump or kayak). I tested with Vector drivers and the update is almost instantaneous. I am wondering what could be causing these delays in socketcan and how can I resolve this issue. Are there any additional settings I will have to do? Appreciate any help 

thank you

hartkopp, 2019-10-26:
Such a delay - especially on candump-level - is very likely no SocketCAN issue.
I assume either the CAN bus is not properly terminated (2x 120 Ohms) or the bitrate settings are not correct (e.g. running at the edge of the specification).
What kind of CAN hardware, Linux distribution and Kernel version are you using?
Did you check the delay without starting socketcand & Kayak?

Prj1508, 2019-10-28:
@hartkopp, thanks for your response. 
I am using combination of these two for CAN hardware
https://comma.ai/shop/products/panda-obd-ii-dongle
https://comma.ai/shop/products/giraffe

I am using these versions
Linux version 4.13.0-37-generic (buildd@lcy01-amd64-012) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.9)) #42~16.04.1-Ubuntu SMP Wed Mar 7 16:03:28 UTC 2018

I did check the delay in candump without starting socketcand and kayak and it is same. THe network is at 500kbps baud rate and hence I have configured my ip link for 500kbps as well
Please let me know your inputs

hartkopp, 2019-10-28:
> I am using combination of these two for CAN hardware
> https://comma.ai/shop/products/panda-obd-ii-dongle

How did you connect the adapter to your Ubuntu machine? Via USB or Wifi?

Prj1508, 2019-10-29:
I have connected via USB
Here is the output:

9: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0 
    can state ERROR-ACTIVE restart-ms 0 
	  bitrate 500000 sample-point 0.000 
	  tq 0 prop-seg 0 phase-seg1 0 phase-seg2 0 sjw 0
	  clock 0

thanks

hartkopp, 2019-10-29:
Ok. I checked the USB SocketCAN driver at
https://github.com/commaai/panda/tree/master/drivers/linux
AFAICS the USB driver just tunnels CAN frames (from up to 3 CAN interfaces) over USB - without any configuration possibilities. So it is no real driver - but just an USB tunnel.
Hm. It's hard to check what the adapter is doing here. Maybe you should better file an issue in the Panda project https://github.com/commaai/panda ?!?

Prj1508, 2019-10-29:
Thanks. I will file a case on Panda page . 
Do you recommend any other HW that works well with SocketCAN and has all termination in place?

Prj1508, 2019-10-29:
I checked with Panda support. It seems https://comma.ai/shop/products/giraffe has 120ohm terminal resistance, but not panda device (https://comma.ai/shop/products/panda-obd-ii-dongle). In the vehicle, I have disconnected one of the ECUs and have physically connected the connector on the vehicle side to these 2 devices which then connects thru USB into my machine. 
Do you think this could be the problem? Should I procure a 120 ohm resistor? I know this is more of a hardware question, but any input is very much appreciated

hartkopp, 2019-10-29:
Yes. Proper termination is the key.
E.g. see here:
https://www.ni.com/en-gb/innovations/white-papers/09/can-physical-layer-and-termination-guide.html
As you can see there are two 120 Ohms resistors in parallel(!) which results in 60 Ohms.
AFAIK the OBD port is already terminated with 60 Ohms between CAN_L and CAN_H.
Therefore the OBD-Adapter needs no termination resistor.
You should check the (internal) resistor of your ECU. E.g. some Engine ECUs already have an internal termination.
So put the stuff together and ensure ~60 Ohms  between CAN_L and CAN_H :-)

josemic, 2020-06-02:
Usually a seperate terminator such as [PCAN Term  or PCAN MiniTerm](https://www.peak-system.com/PCAN-Term.270.0.html?&L=1) or self build is used with CAN-adapters.