linux-can/can-utils #161: fatal error: error.h: No such file or directory

brandonros, 2019-10-30:
```
~/can-utils # apk update && apk upgrade && apk add --no-cache git make build-base linux-headers gcc libc-dev cm
ake
fetch http://dl-cdn.alpinelinux.org/alpine/v3.10/main/x86_64/APKINDEX.tar.gz
fetch http://dl-cdn.alpinelinux.org/alpine/v3.10/community/x86_64/APKINDEX.tar.gz
v3.10.3-11-g33bd61f256 [http://dl-cdn.alpinelinux.org/alpine/v3.10/main]
v3.10.3-13-g8068beb776 [http://dl-cdn.alpinelinux.org/alpine/v3.10/community]
OK: 10336 distinct packages available
OK: 185 MiB in 39 packages
fetch http://dl-cdn.alpinelinux.org/alpine/v3.10/main/x86_64/APKINDEX.tar.gz
fetch http://dl-cdn.alpinelinux.org/alpine/v3.10/community/x86_64/APKINDEX.tar.gz
(1/11) Installing libacl (2.2.52-r6)
(2/11) Installing libbz2 (1.0.6-r7)
(3/11) Installing lz4-libs (1.9.1-r0)
(4/11) Installing xz-libs (5.2.4-r0)
(5/11) Installing libarchive (3.3.3-r0)
(6/11) Installing ncurses-terminfo-base (6.1_p20190518-r0)
(7/11) Installing ncurses-terminfo (6.1_p20190518-r0)
(8/11) Installing ncurses-libs (6.1_p20190518-r0)
(9/11) Installing rhash-libs (1.3.8-r0)
(10/11) Installing libuv (1.29.1-r0)
(11/11) Installing cmake (3.14.5-r0)
Executing busybox-1.30.1-r3.trigger
OK: 221 MiB in 50 packages
~/can-utils # cmake
Usage

  cmake [options] <path-to-source>
  cmake [options] <path-to-existing-build>
  cmake [options] -S <path-to-source> -B <path-to-build>

Specify a source directory to (re-)generate a build system for it in the
current working directory.  Specify an existing build directory to
re-generate its build system.

Run 'cmake --help' for more information.

~/can-utils # cmake .
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /root/can-utils
~/can-utils # make
Scanning dependencies of target can-calc-bit-timing
[  1%] Building C object CMakeFiles/can-calc-bit-timing.dir/can-calc-bit-timing.c.o
/root/can-utils/can-calc-bit-timing.c: In function 'print_bit_timing':
/root/can-utils/can-calc-bit-timing.c:538:73: warning: 'tseg1' may be used uninitialized in this function [-Wmaybe-uninitialized]
  bt->bitrate = priv->clock.freq / (bt->brp * (CAN_CALC_SYNC_SEG + tseg1 + tseg2));
                                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/root/can-utils/can-calc-bit-timing.c:447:35: note: 'tseg1' was declared here
  unsigned int brp, tsegall, tseg, tseg1, tseg2;
                                   ^~~~~
/root/can-utils/can-calc-bit-timing.c:532:12: warning: 'tseg2' may be used uninitialized in this function [-Wmaybe-uninitialized]
    bt->sjw = tseg2;
    ~~~~~~~~^~~~~~~
/root/can-utils/can-calc-bit-timing.c:447:42: note: 'tseg2' was declared here
  unsigned int brp, tsegall, tseg, tseg1, tseg2;
                                          ^~~~~
[  3%] Linking C executable can-calc-bit-timing
[  3%] Built target can-calc-bit-timing
Scanning dependencies of target can
[  4%] Building C object CMakeFiles/can.dir/lib.c.o
[  6%] Building C object CMakeFiles/can.dir/canframelen.c.o
[  7%] Linking C static library libcan.a
[  7%] Built target can
Scanning dependencies of target canlogserver
[  9%] Building C object CMakeFiles/canlogserver.dir/canlogserver.c.o
[ 11%] Linking C executable canlogserver
[ 11%] Built target canlogserver
Scanning dependencies of target bcmserver
[ 12%] Building C object CMakeFiles/bcmserver.dir/bcmserver.c.o
[ 14%] Linking C executable bcmserver
[ 14%] Built target bcmserver
Scanning dependencies of target j1939
[ 15%] Building C object CMakeFiles/j1939.dir/libj1939.c.o
/root/can-utils/libj1939.c:19:10: fatal error: error.h: No such file or directory
 #include <error.h>
          ^~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/j1939.dir/build.make:63: CMakeFiles/j1939.dir/libj1939.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:184: CMakeFiles/j1939.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

brandonros, 2019-10-30:
```
  33 apk update && apk upgrade && apk add --no-cache git make build-base linux-headers gcc libc-dev cmake ninja autoconf automake libtool
  34 sh autogen.sh 
  35 ./configure
  36 make
```

does not work either

```
  CC       jacd.o
jacd.c:23:10: fatal error: error.h: No such file or directory
 #include <error.h>
          ^~~~~~~~~
compilation terminated.
make: *** [GNUmakefile:819: jacd.o] Error 1
```

marckleinebudde, 2019-10-30:
I have a `/usr/include/error.h` on debian system. It's part of: `libc6-dev:amd64: /usr/include/error.h`

marckleinebudde, 2019-10-30:
We just noticed that you're on alpine linux which uses libmusl as the system C library. It's probably missing the `error.h` header.

http://man7.org/linux/man-pages/man3/error.3.html

marckleinebudde, 2019-10-30:
Can you test this branch:

https://github.com/marckleinebudde/can-utils/tree/libmusl-fix

marckleinebudde, 2019-10-31:
Yes, the glibc does also support err.h. I'll create a pull request.

Tnx,
Marc

closed 2019-10-31 by marckleinebudde