linux-can/can-utils #170: candump can0 does not display CAN data

Prj1508, 2019-11-17:
I am using PEAK-USB Interface in my vehicle and I am running ubuntu 16.04 in my machine.  I am able to add the device, set the link up for can0, I have set the bitrate etc., but candump can0 is not displaying the CAN traffic. The output of ip -det link show can0 is

4: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0 
    can state ERROR-ACTIVE restart-ms 0 
	  bitrate 500000 sample-point 0.875 
	  tq 125 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
	  pcan_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 8000000
Appreciate your help
thanks

hartkopp, 2019-11-17:
The PEAK-USB adapters have a status LED which is blinking in a low frequency (~1 Hz) when the interface is set to "up". When the adapter sends/receives CAN frames the blink frequency increases (~2 Hz).
Can you see the LED speeding up its blink frequency when connecting to the CAN bus with CAN traffic?

Prj1508, 2019-11-18:
@hartkopp, thanks. 
it keeps flashing at 1Hz and doesn’t speed up when connected to a real CAN bus.

hartkopp, 2019-11-18:
I assume you have not connected the CAN bus properly to the existing CAN network you want to monitor. E.g. swapped CAN_L / CAN_H ?
What kind of setup did you connect the CAN-USB adapter to? A car?

Prj1508, 2019-11-18:
i am using OBD to serial link cable to connect from Car to PEAK and then thru USB to connect my laptop

hartkopp, 2019-11-18:
Do you have an URL for the USB to serial cable?

Prj1508, 2019-11-18:
yes
https://www.wantitall.co.za/automotive/fotag-obd2-16-pin-to-db9-rs232-serial-port-adapter-cable-diagnostic-extension-cable-with-power-switc__b01etrinyo

Prj1508, 2019-11-18:
Power switch is ON as well

hartkopp, 2019-11-18:
This adapter cable does not work for you!
The OBD port has CAN_L on pin 14 and CAN_H on pin 6.
The CAN CiA Standard SUB-D 9 (which is used by PEAK) is CAN_L on pin 2 and CAN_H on pin 7.
But your adapter cable puts CAN_L (pin 14) to pin 5 on the SUB-D 9 and CAN_H (pin 6) to pin 3 on the SUB-D 9 connector. Completely wrong!
Hopefully you did not kill your CAN USB adapter. When it is unmodified the +12V on pin 9 of the SUB-D 9 is not connected inside the adapter. See pin assignment:
https://www.peak-system.com/PCAN-USB.199.0.html?&L=1

Prj1508, 2019-11-18:
Thank you. 
Appreciate if you can recommend any other make of adapter cable that is a plug-n-play for PEAK CAN USB adapter

hartkopp, 2019-11-18:
https://www.google.com/search?client=firefox-b-e&q=CAN-OBD-2+sub-d+can+cia
This leads also to a PEAK product which only connects

OBD Pin 14 -> SUB-D Pin 2 (CAN_L)
OBD Pin 6 -> SUB-D Pin 7 (CAN_H)

I would suggest to cut your current (incorrect) cable in two pieces and wire only these two lines. Or you add another SUB-D 9 (male) to SUB-D 9 (female) adapter which only swaps the required lines as required.

Prj1508, 2019-11-18:
Thank you so much

hartkopp, 2019-11-18:
:-)

closed 2019-11-18 by hartkopp

marckleinebudde, 2021-06-02:
please open a new issue. Thanks

monajalal, 2024-09-18:
> can someone help out with this similiar problem?
> 
> We have a Nvidia Jetson Nano (UBUNTU 18.04.LTS) connected to a USB-CAN Adapter (Peak). The Adapter is linked via CAN to a CVMU (Cell Voltage Monitoring Unit) to observe the voltages of a small fuell cell stack. We set up the adapter by typing the following into the terminal:
> 
> 1. $ sudo apt-get install can-utils
> 2. $ sudo ip link set can0 up type can bitrate 500000
> 3. $ sudo ip link set can0 up            (Adapter control LED signals that it is now connected (1Hz blinking))
> 4. $ ifconifg  (can0 is displayed)
> 
> When we now command in the terminal $ candump can0 the curser jumps on the next line (no more directory) and nothing happens anymore. Also the Adapter LED does not signal data transfer (should be 2Hz blinking) We have been not able to find out if this is related to a software or hardware issue. If someone can give us useful hints, we would appreciate it! Question is, if the terminal shows nothing because there is no data transfer and the cables have to be checked or if there are further interface configurations in Linux necessary. Kind regards

I have the same issue on Jetson AGX ORIN 64G DevKit. Did you find a solution?

hartkopp, 2024-09-18:
When the bitrate is surely 500000 bit/s then it is very likely a hardware issue:
- no connection
- no termination (must be 60 Ohms == 2 x 120 Ohms)
- no traffic from the sender side