linux-can/can-utils #171: j1939 on BeagleBone

GwenStouthuysen, 2019-11-21:
I'm stuck with J1939 on Beaglebone.
uname -a: Linux beaglebone 4.19.79-ti-r30 #1stretch SMP PREEMPT Mon Nov 4 20:48:10 UTC 2019 armv7l GNU/Linux

Installed the last status of can-utils from this repository (./autogen.sh till make install)

I have a functional CAN socket with J1939 equipment sending messages:
debian@beaglebone:~/can-utils$ candump can0
  can0  18EF3C33   [8]  20 54 F8 33 12 00 02 07
  can0  18EF3C33   [8]  30 BF 74 67 EC 00 00 0E
  can0  18EF3C33   [8]  30 BF 74 67 EC 00 00 0D

Trying to review this with a J1939 enabled socket fails:
debian@beaglebone:~/can-utils$ testj1939 -r -v can0
- socket(PF_CAN, SOCK_DGRAM, CAN_J1939);
testj1939: socket(j1939): Invalid argument

What do I do wrong?

yegorich, 2019-11-21:
You'll have to build/install at least kernel 5.4-rc8.

GwenStouthuysen, 2019-11-28:
And can you explain how to do this?

I tried to upgrade the kernel, using the standard script and apt, when I go
fro 4.x to 5.x the unit does not boot anymore.
I didn't go that far yet to capture the boor process but simply said: it
doesn't work.

Gwen

yegorich, 2019-11-28:
You can try [Buildroot](https://buildroot.org/). It already provides a defconfig for different BeagleBones (`configs/beaglebone_defconfig`).

```
git clone git://git.buildroot.net/buildroot
cd buildroot
make beaglebone_defconfig
make menuconfig
```

Once in menuconfig, change Kernel configuration to just use the upstream kernel and provide an URL to [linux-5.4](https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.4.tar.xz) and also select iproute2 and can-utils. Save & exit.

```
make
make linux-menuconfig
```

Check whether J1939 is enabled and repeat `make`.

Now you should have an SD card image with the right kernel and configuration.

Once you're satisfied with kernel configuration, you can copy kernel and modules to your Debian card and use it there.