linux-can/can-utils #173: Networking with CAN-J1939 support (Ubuntu Linux) jacd.c usage

josemic, 2019-11-27:
This is a cleanup of part of the discussion in #159. 
Using Ubuntu linux with kernel 5.4. The installation is described here: [link](http://www.theubuntumaniac.com/2019/10/upgrade-to-linux-kernel-54-rc1-on.html) on Ubuntu 19.10.
I have deinstalled the ubuntu can-utils package with "sudo apt remove can-utils".
I use latest can-utils from git.
Setup:
```
sudo modprobe vcan
sudo ip link add can0 type vcan
sudo ip link set can0 up
sudo modprobe can-j1939
```

I have 4 findings:
1. jacd has to be called with sudo. Why? I think this should not be the case. I especially will have problems to run TTCN test cases in this case or when the sochetCAN socket commands are used in other programs. In worst case I have to change root access rights with visudo.
2. when the command is called with an "&" at the end of the line, the command does not work, as the password is not queried, as the command is running in background. This will not query for an password:
`sudo ./jacd -r 100,80-120 -c /tmp/1122334455667781.jacd 1122334455667781 can0 &`
,but this without the "&" at the end will query for a password:
`sudo ./jacd -r 100,80-120 -c /tmp/1122334455667781.jacd 1122334455667781 can0`
3. when using the sudo javd with "-v" option and without "&" the command fails with the message "No such file or directory". I really do not understand why this is happening. Verbosity should not have any impact on the command outcome.
```
sudo ./jacd -r 100,80-120 -v -c /tmp/1122334455667781.jacd 1122334455667781 can0
jacd: ready for can0:1122334455667781: No such file or directory
```
Sometimes also "success" is printed, but noting else and no candump output is generated. Also the prompt returns immediately. (This seems to happen, if jacd had been called before once without "-v" option):

On Termial 1:
```
sudo ./jacd -r 100,80-120 -v -c /tmp/1122334455667780.jacd 1122334455667780 can0
jacd: ready for can0:1122334455667780: Success
```
On Terminal 2:
```
sudo ./candump -t d can0
<EMPTY>
```
4. when using the sudo jacd without "-v" option and without "&" the command works:

On Terminal 1:
`sudo ./jacd -r 100,80-120 -c /tmp/1122334455667781.jacd 1122334455667781 can0`

On Terminal 2:
```
sudo ./candump -t d can0
 (000.000000)  can0  18EAFFFE   [3]  00 EE 00
 (000.000195)  can0  18EEFF50   [8]  81 77 66 55 44 33 22 11
 (001.250191)  can0  18EEFF50   [8]  81 77 66 55 44 33 22 11
```
And after ctrl-c in Terminal 1:

On Terminal 2:
`(212.451499)  can0  18EEFFFE   [8]  81 77 66 55 44 33 22 11`

olerem, 2019-12-06:
This issue should be fixed by this pull request: https://github.com/linux-can/can-utils/pull/180

marckleinebudde, 2019-12-06:
Regarding `sudo`: If you have `sudo` configured to ask for a password, you cannot start it in the background (using `&`). This is normal `sudo` behaviour, that has nothing todo with `j1939` or `jacd`.

josemic, 2019-12-10:
The 'sudo' is now no longer necessary. That is working. 

What is still strange is that the -v option is not working. 

In Terminal 1:
`./candump -t d can0`
  
In Terminal 2:
`./jacd -r 100,80-120 -v -c /tmp/1122334455667780.jacd 1122334455667780 can0`

Result in Terminal 2:
`jacd: ready for can0:1122334455667780: Success`

or sometimes:
`jacd: ready for vcan0:1122334455667780: No such file or directory`

in both cases in Terminal 1:
`<EMPTY>`

What is working, when jacd is called without '-v' option:

In Terminal 2:
`./jacd -r 100,80-120 -c /tmp/1122334455667780.jacd 1122334455667780 can0`

In  Terminal 1:
```
./candump -t d can0
 (000.000000)  can0  18EAFFFE   [3]  00 EE 00
 (001.250231)  can0  18EEFF50   [8]  80 77 66 55 44 33 22 11
```
And after ctrl-C in Terminal 2 the Terminal 1 shows:
` (104.651817)  can0  18EEFFFE   [8]  80 77 66 55 44 33 22 11`

```

josemic, 2020-05-26:
for the record: jacd had been renamed to j1939acd in latest can-utils

josemic, 2020-05-26:
I this issue is closed. I have created as follow-up #206 .

closed 2020-05-26 by josemic