linux-can/can-utils #179: Networking with CAN-J1939 - remove root requirements

josemic, 2019-12-02:
There seem to be some commands (setsockopt, sendto) in some cases required for dynamic addressing which require  root access. I am not aware that j1939 dynamic addressing is more critical or harmful than static addressing. 
Thus please remove root requirements for dynamic addressing.

marckleinebudde, 2019-12-06:
Do you have an example where you need root privileges?

josemic, 2019-12-10:
This was referring to jacd usage in #173. Which is now fixed. I had when using dynamic addressing in TTCN simialr problems, but that seems to be caused by the same "SO_BINDTODEVICE setsockopt" call copied from jacd into the TTCN code. I think you can close the issue now, as the root reuirements in #173 have been fixed.

closed 2019-12-10 by marckleinebudde