linux-can/can-utils #181: make the cansniffer 29bit ID capable

marckleinebudde, 2019-12-06:


marckleinebudde, 2019-12-06:
See https://github.com/linux-can/can-utils/issues/71#issuecomment-408702680 and following.

hartkopp, 2020-04-26:
Hi @coryjfowler @josemic @Swompet 
Corona boredom and https://virtual.carhackingvillage.com/ CTF made me think about the 29bit support again (and again). My approach to support multiple CAN interfaces and CAN FD from the beginning was too complex. So I reduced my current rework on these topics and adopted the slot concept for the BCM->RAW conversion from  @coryjfowler https://github.com/coryjfowler/can-utils which really helped to start over.

Before starting the CAN FD support I would like to invite you to test the current rework and hopefully give some feedback.

You can find it in the "snif" branch of my can-utils fork here:
https://github.com/hartkopp/can-utils/
https://github.com/hartkopp/can-utils/tree/snif

Thanks,
Oliver

geier99, 2020-04-28:
Hi Oliver,
It's nice to see that you worked on it.
I did some tests on an raspberry pi 3 B and it's working very well.

The only thing I don't like is that you have removed the filter function with "CAN-ID and Bitmask".
(for example: "+100700"  setting a filter ID range from 0x100 til 0x1FF)

Setting filters for a single CAN-ID works also well.

I hope you see a way to support the filter masks again because I've used them a lot in the past.

Thanks a lot,
Andreas

hartkopp, 2020-04-28:
Hi Andreas,
the filter has been removed by @coryjfowler in the conversion process as it was some kind of tricky to apply it for EFF frames :-)
Btw. I implemented the filters for SFF and also EFF again.
And we now have the possibility to enable and disable EFF/SFF CAN ID as a whole.
Just check out the latest code and see the updated help text.

Many thanks for showing up here! Usually I do not get any feedback which functionalities are really need by the users out there ;-)

josemic, 2020-04-30:
Hi Oliver, 
I wanted to use it for J1939 support, however I noticed that candump does is better suited for the job. What would be even better suited would be a j1939dump, which dumps aggregated j1939 frames instead of can frames.
Michael

hartkopp, 2020-04-30:
Ah, ok. Then this would be more a request for @olerem who is working on the j1939 stack & tools ;-)

josemic, 2020-04-30:
Yes. It is just an idea.

hartkopp, 2020-05-07:
Mission accomplished by commit https://github.com/linux-can/can-utils/commit/c5cb7e1a7c0562dd52c295583d859c5920abbbb7

closed 2020-05-07 by hartkopp

olerem, 2020-05-08:
@josemic , we already have jspy (j1939spy). The only challenge, is the nature of J1939/IsoBUS transfer protocol. It happens between two nodes and if the third node is spying, it should only assemble the package without interacting with actual communicating nodes.
It should be possible to implement it, but currently it has no priority for me. If you can describe, why it is the best and most important future every one ever wonted, then may be we will be able to push it forward sooner

josemic, 2020-05-22:
@olerem, ok. I have run j1939spy. It works, my main issue is that it does not reassembly of segmented j1939 messages (this applies especially for the PGNs 00EB00 and 00EC00). This is something I would expect from j1939dump. Perhaps I am wrong.
I have here a log file, I have taken some years ago from an Isobus device with candump. I played it with 'canplayer -I mylog.log ' and take a trace with j1939spy.
Also I have here attached as the same replayed reassembled with './testj1939 can0:0xF8 -r &'.

[mylog.log](https://github.com/linux-can/can-utils/files/4667569/mylog.log)
[j1939spy.log](https://github.com/linux-can/can-utils/files/4667563/j1939spy.log)
[mylog_can0_canaddr_F8_reassembled.log](https://github.com/linux-can/can-utils/files/4667630/mylog_can0_canaddr_F8_reassembled.log)

olerem, 2020-05-22:
@josemic , yes, this functionality is currently not supported. I opened an issue to track it: https://github.com/linux-can/linux/issues/6