linux-can/can-utils #184: Doubt:29 bit identifier

srigandeswaran, 2019-12-13:
Hi all,
 I have a doubt that,whether the candump.c and cansend.c code can be used for SW CAN also and whether it will be suitable for 29 bit identifier.

could anyone please answer?
Thanks in advance.

hartkopp, 2019-12-13:
Hi,
1. what do you mean with "SW CAN"?
2. where is the problem with 29 bit IDs?

`cansend vcan0 12345678#112233`

and in another terminal you can then see

`$ candump any`
`  vcan0  12345678   [3]  11 22 33`

Looks good to me.

olerem, 2019-12-13:
Hm... single wire CAN?
https://www.nxp.com/products/interfaces/can-transceivers/legacy-can/single-wire-can-transceiver:MC33897

hartkopp, 2019-12-13:
This is a transceiver (= physical layer) which is connected to a CAN controller (which usually supports 29 bit IDs).
Is question "2" answered then?

srigandeswaran, 2019-12-13:
@hartkopp 
The question is whether these **candump.c** can receive the CAN ID's with 29bit identifier?
it is receiving the 11 bit identifier CAN id's.

hartkopp, 2019-12-13:
Yes it can - as you can see above.
In candump and cansend 29 bit IDs are represented with 8 ASCII digits (in the example 12345678) and 11 bit IDs  are represented with 3 ASCII digits (e.g. 3A4 or 123)

olerem, 2019-12-13:
11 bit = 11111111111b = 0x7FF
29 bit = 11111111111111111111111111111b = 0x1FFFFFFF

srigandeswaran, 2019-12-13:
@hartkopp 
ok thank you,I will check these out.

closed 2020-01-31 by hartkopp