linux-can/can-utils #186: Not able to publish on  CAN bus

Prj1508, 2019-12-17:
Hi

I am trying to publish a CAN message at 100ms rate on to a topic using a timercallback function. I am running the code onboard a vehicle and I have a Ubuntu 16.04 machine with ROS Kinetic and has ros-canopen/socketcan_bridge package. I am using PEAK USB-CAN Interface. When I publish the message on sent_message topic, it is not being sent to CAN bus and ROS gives out an error message "Failed to send message.....".

I suspected this could be due to CAN driver not being ready and I also see that the CPU/processor usage spikes (uses up about 65% of processor time) when I run the node that does this publishing. My code is straightforward in the callback and should not typically consume lot of resources ( I have i9-6700HQ processor, 16GB RAM with GTX1070). Here is the error message. 

ERROR] [1576620411.175770486]: Failed to send message: 292#57ff0000c0f90000.
[ERROR] [1576620411.175776441]: Failed to send message: 292#57ff0000d0340000.
[ERROR] [1576620411.175803531]: Failed to send message: 292#57ff0000d0340000.
[ERROR] [1576620411.175809256]: Failed to send message: 292#57ff0000e07e0000.
[ERROR] [1576620411.175836518]: Failed to send message: 292#57ff0000e07e0000.
[ERROR] [1576620411.175843258]: Failed to send message: 292#57ff0000f0b30000.
[ERROR] [1576620411.175870902]: Failed to send message: 292#57ff0000f0b30000.
[ERROR] [1576620411.175875283]: Failed to send message: 292#57ff000000cc0000.
[ERROR] [1576620411.175903887]: Failed to send message: 292#57ff000000cc0000.
[ERROR] [1576620411.175908951]: Failed to send message: 292#57ff000010010000.
[ERROR] [1576620411.175937795]: Failed to send message: 292#57ff00001001000

Can you please help?

 Thank you

hartkopp, 2019-12-18:
That's a pretty long chain of dependencies.
Did you try whether the CAN bus interface works and is able to send CAN frames? E.g. with cansend and cangen tools?

Prj1508, 2019-12-18:
Yes, i did and cansend works fine. The problem is mostly when i send thru a timercallback @ 100ms rate

hartkopp, 2019-12-18:
what happens with
`cangen -g 100 can0`
which also creates CAN traffic with the same 100ms rate?
(you can also specify the content of the CAN frame - see `cangen -h`)

Prj1508, 2019-12-18:
i tried cangen -g 100 can0 and it goes thru just fine

hartkopp, 2019-12-20:
Well. Then the problem resides somewhere in ROS or the ros-canopen/socketcan_bridge package ...
Maybe you should ask for help in the ROS / ros-canopen community?!?

closed 2020-01-31 by hartkopp