linux-can/can-utils #187: New release?

Bastian-Krause, 2019-12-19:
Hi,

the last (and only) release was quite a long time ago. It would be nice to have a more recent one which could then end up in a recent Debian package.

Regards,
Bastian

a3f, 2020-01-06:
Preferably, this release could include the PTX flavour can-utils like cansequence as well!

hartkopp, 2020-01-31:
I would also vote for a new tag v2020.02.0
Especially due to the fact, that the code is stored in a mountain in some day :-)

hartkopp, 2020-02-01:
And there we are with a new tag!
https://github.com/linux-can/can-utils/releases/tag/v2020.02.0
Thanks to @Bastian-Krause for the reminder!
For the next step/tag we could hopefully really enable cansniffer for 29bit IDs and CAN FD - and to create man pages for the tools. :-D

closed 2020-02-01 by hartkopp