linux-can/can-utils #188: candump empty

stapperfend, 2019-12-20:
Hello, the candump for us is empty when we try to read out can0. Reading out vcan works fine using the same setup and commands. We use canopensocket

hartkopp, 2019-12-20:
Hi. I would assume that you have not set up the can0 interface properly (e.g. bitrate, setting the interface to 'up') or that you don't have any traffic on the bus or a missing CAN bus termination.
When candump works on vcan0 it should be able to read from can0 in the same way.

stapperfend, 2019-12-20:
We used it before

hartkopp, 2019-12-20:
You used what?

stapperfend, 2019-12-20:
> You used what?

Didn't mean to send the response. Sorry. We used candump before. So we thought our issue might be with an update of can_utils.

hartkopp, 2019-12-20:
You are using the mttcan which is a NVIDIA out of tree CAN driver. I would suggest to look into that direction (and probably the sampling-point option).
The can-utils (and especially candump) should not have changed in a way that it has an effect to your updated setup.

stapperfend, 2019-12-20:
okay. Thank you a lot for your time.
That helps us isolating the error.

closed 2019-12-20 by stapperfend