linux-can/can-utils #190: Cannot send 29 bit extended CAN messages

VIMALNATHGM, 2019-12-24:
Hi,
I'm using a SW CAN(Single wire CAN) with extended CAN id's. I have used candump to receive the extended CAN messages, but I can't transmit any messages. I have tried with some formats but I'm not able to get proper data's. Can anyone tell me the format to send extended CAN messages.

olerem, 2019-12-24:
Please, tell a bit more hardware. SW CAN(Single wire CAN) is not hardware. It is even not official standard.

try: cangen -v can0
Instead of can0 use the interface name on your system.

VIMALNATHGM, 2019-12-24:
Hi,
I'm trying to send this command > cansend can0 1f123456#1122334455667788, but I'm getting this in my CAN analyser tool >ID=1, Length=1, Databytes=1.

VIMALNATHGM, 2019-12-27:
@olerem I'm using NXP i.mxUL SOM and NCV 7356-D CAN transceiver and linux OS.

willianaraujo, 2019-12-27:
> Hi,
> I'm trying to send this command > cansend can0 1f123456#1122334455667788, but I'm getting this in my CAN analyser tool >ID=1, Length=1, Databytes=1.
try:
`cansend can0 -e -i 521286742 11 22 33 44 55 66 77 88`

VIMALNATHGM, 2020-01-04:
@willianaraujo Thanks for your suggestion. Now it's working fine. I tried this format->
'cansend can0 -e -i 0x1f345678 0x11 0x22 0x33 0x44 0x55 0x66 0x77 0x88'. It's taking only in hex format.

hartkopp, 2020-01-04:
We should make clear that 
> 'cansend can0 -e -i 0x1f345678 0x11 0x22 0x33 0x44 0x55 0x66 0x77 0x88'. It's taking only in hex format.

**ONLY** works with the **PENGUTRONIX FLAVOUR** of **canutils** that can be found in some BSPs and here: https://git.pengutronix.de/cgit/tools/canutils/

BUT THIS issue has been filed and discussed for **can-utils** (with a **"-"**) on GitHub (https://github.com/linux-can/can-utils) where only `cansend can0 1F345678#1122334455667788` would be correct. 

**Please do not mix these packages (confusing readers) and preferably ask on the Pengutronix mailing list when having issues with that package.**
Thanks!

closed 2020-01-04 by hartkopp