linux-can/can-utils #197: Can J1939 on Kernels lower than 5.4

srsidd, 2020-02-28:
Hi,

We have a motor drive that runs CAN J1939 we need to communicate with. We're on Ubuntu 18.04 with a kernel version `5.0.0-23-generic`. I tried following the [J1939 Kickstarter guide](https://github.com/linux-can/can-utils/blob/master/can-j1939-kickstart.md) but ran into an error with setting up the `can-j1939` module - 

```
foo@bar ~$ sudo modprobe can-j1939
modprobe: FATAL: Module can-j1939 not found in directory /lib/modules/5.0.0-23-generic
```

I did some googling, and according to the [Elinux wiki](https://elinux.org/J1939), 

> SAE J1939 got in mainline kernel since v5.4 .

I've tried searching far and wide, but am still not sure how to get this on the kernel we have. Updating the kernel is not an option, do you have any pointers on how I can get `modprobe` to recognize the `can-j1939` module?

marckleinebudde, 2020-03-02:
Hey @srsidd,

I personally don't have any motivation to port the j1939 stack to old kernels as community efford. Yes the stack needs testing, but the automated testing found several issues in the initial commit. The maintenance cost of taking care of j1939 in old kernels (that are end-of-life by kernel.org anyways) is just too high. If you need commercial support drop us a note.

srsidd, 2020-03-02:
I'm a little lost. I was of the opinion that the J1939 is _not_ that recent. How were people using it in on Ubuntu till now? Is there a `src` build I can try?

On a slightly different note, what is the effort involved in backporting to an older kernel? Maybe that's something you're open to a PR if someone submits one?

marckleinebudde, 2020-03-02:
> I'm a little lost. I was of the opinion that the J1939 is _not_ that recent.

It was recently merged to the kernel with a stable API. Everything else before this is obsolete.

> How were people using it in on Ubuntu till now? Is there a `src` build I can try?

J1939 is in the mainline kernel.

> On a slightly different note, what is the effort involved in backporting to an older kernel?

Backport the `j1939` stack, backport the fixes, backport the CAN stack changes that are needed for j1939 support, backport driver infrastructure fixes and backport driver fixes.

> Maybe that's something you're open to a PR if someone submits one?

Sorry, as a community effort, I personally will not support the j1939 stack in kernels older than v5.4.

srsidd, 2020-03-05:
What I'm hearing is even if I were to try a manual src build or binary on a lower kernel, I'd run into issues because it's buggy\obsolete\incompatible and it won't be any easy task to resolve those. 

Ok, we can close this issue unless you think it's still a valid issue and will get resolved sometime in the future. I think I got my question answered.

closed 2020-03-05 by srsidd

marckleinebudde, 2020-03-05:
Updating for your kernel is an investment you have to do anyways, sooner or later. Backporting j1939 to an old kernel isn't.

zeljkoavramovic, 2020-09-30:
I am sorry to bring this up again, but I have the same problem on kernel 5.7:

> avra@vm-debian:~/can/can-utils$ sudo modprobe can-j1939
> modprobe: FATAL: Module can-j1939 not found in directory /lib/modules/5.7.0.0.bpo.2-amd64

> avra@vm-debian:~/can/can-utils$ sudo testj1939
> testj1939: socket(j1939): Protocol not supported

Searching files on my system I can find can.ko, can-raw.ko, vcan.ko but nothing similar to *1939*.ko.

My virtual machine is Debian 10.5 testing with kernel 5.7 from backports:
> avra@vm-debian:~/can/can-utils$ uname -a
> Linux vm-debian 5.7.0.0.bpo.2-amd64 #1 SMP Debian 5.7.10-1-bpo10+1 (2020-07-30) x86_64 GNU/Linux

vcan0 is up and working, with sending and receiving frames without problems. Kayak is also working well over socketcand.

Kickstart guide fails at modprobe can-j1939 so I am stuck. Could you please give me a pointer on what could I try next?