linux-can/can-utils #198: Please clarify license and copyright for generic files

gerasiov, 2020-04-11:
Most files in repository have legal information in its headers, but not all of them.

> W: can-utils source: file-without-copyright-information .github/workflows/autotools.yml
> W: can-utils source: file-without-copyright-information .gitignore
> W: can-utils source: file-without-copyright-information .travis.yml
> W: can-utils source: file-without-copyright-information Android.mk
> W: can-utils source: file-without-copyright-information CMakeLists.txt
> W: can-utils source: file-without-copyright-information GNUmakefile.am
> W: can-utils source: file-without-copyright-information README.md
> W: can-utils source: file-without-copyright-information autogen.sh
> W: can-utils source: file-without-copyright-information cmake/make_uninstall.cmake
> W: can-utils source: file-without-copyright-information config/m4/.secret-world-domination-project
> W: can-utils source: file-without-copyright-information configure.ac
> W: can-utils source: file-without-copyright-information page.theme
> W: can-utils source: file-without-copyright-information style.css

I thinks it would be better to set (with COPYRIGHT|LICENSE|AUTHORS file or any similar way) some kind of "default" license for the repo.

hartkopp, 2020-04-25:
Hello @gerasiov,
we already have the licenses in the LICENSES directory.
Don't you think this is enough? Or do you suggest to put extra SPDX identifiers in each of the above shown files?

yegorich, 2020-04-25:
If you look at the reuse example [repository](https://github.com/fsfe/reuse-example), you can see that every file has a license. Even `README.md` and `.gitignore`. What about adding SPDX identifiers (GPL-2.0-only) for all these files?

hartkopp, 2020-04-26:
README.md seems to have no license - and some of the files (especially text/documentation) have Creative-Commons licenses.
Maybe this could be an approach to put missing Makefiles/code stuff under GPL-2.0-only and the rest under some permissive CC license?!?

yegorich, 2020-04-27:
@hartkopp I like this approach. 

BTW `README.md` does have a license. You can see it only if you open the file as raw.

```
<!--
SPDX-FileCopyrightText: 2019 Jane Doe <jane@example.com>

SPDX-License-Identifier: GPL-3.0-or-later
-->
```

gerasiov, 2020-04-27:
> 2019 Jane Doe <jane@example.com>

hartkopp, 2020-05-07:
@marckleinebudde can you please take over here?
It's mostly cmake/Android build stuff and of course the .secret-world-domination-project where I have no clue where it comes from and what kind of license could be appropriate.

hartkopp, 2020-12-02:
Hi all,
I would like to pick this topic up again, as we had many updates and fixes since out last tag.
It would be cool if @gerasiov could upstream the latest can-utils until Debian 11 (bullseye) is to be released on 2021-01-12 !

nullr0ute, 2025-01-29:
It would be useful to have a file, or details in the readme, that outlines the general overall license, it's not particularly clear from the license files dumped in LICENSES/

gerasiov, 2025-01-29:
For Debian distribution we have the following summary file:
https://sources.debian.org/src/can-utils/2023.03-1/debian/copyright/
But it does not specify resulting summary for binaries.

hartkopp, 2025-01-30:
Hi @gerasiov !
Nice to read from you!
I tagged a new release yesterday.
Maybe this is something for a Debian can-utils package update too?

yegorich, 2025-02-03:
Here is an interesting example for a Debian package using REUSE: https://salsa.debian.org/games-team/game-data-packager/-/commit/ddf3167a21851b363ef8d3a783ca8a7272b926cd

The `COPYING` file is now very short:

> This project's licensing is REUSE-compliant <https://reuse.software/>.
> See individual files for full details of copyright and licensing,
> and see LICENSES/*.txt for the license text.

This info can be added to our `README.md` file.