linux-can/can-utils #202: Android.mk out of date?

hartkopp, 2020-05-07:
As Travis-CI recently detected a problem doing an Android NDK build I wondered why Android.mk had its last update in 2015 - while there have been several new tools (e.g. for j1939) added since then.
Is Android.mk still relevant? Should it been updated?

gibsson, 2020-05-28:
For AOSP build, you'll have to switch to Android.bp (new format) at some point:
https://source.android.com/setup/build#androidbp_file_format
I don't really use NDK so I can't say for sure what's its requirement.

hartkopp, 2020-05-28:
But how are you building your code right now? Does it build the J1939 stuff which has been added at last?

gibsson, 2020-05-28:
I'm building the AOSP way, so either by defining the tools I want as part of `PRODUCT_PACKAGES` or by forcing the build as follows:
```
$ mmm external/can-utils/
```
And no j1939 isn't built as it hasn't been added to `Android.mk`.

hartkopp, 2020-05-29:
Created pull request to fix this issue https://github.com/linux-can/can-utils/pull/215

closed 2020-06-04 by hartkopp