linux-can/can-utils #206: j1939acd -v option not working

josemic, 2020-05-26:
This is a cleanup of part of the discussion in #173. 

```
uname -a
Linux michael-ThinkPad-T430 5.4.0-31-lowlatency #35-Ubuntu SMP PREEMPT Thu May 7 22:51:24 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

I have deinstalled the ubuntu can-utils package with "sudo apt remove can-utils" on Ubuntu 20.04.
I use latest can-utils from git.
Setup:
```
sudo modprobe vcan
sudo ip link add can0 type vcan
sudo ip link set can0 up
sudo modprobe can-j1939
```
On Terminal 1:

`./candump -t d can0`

On Terminal 2:
`j1939acd -r 100,80-120 -c /tmp/1122334455667788.jacd 1122334455667788`

Result on Terminal 1:
```
 (000.000000)  can0  18EAFFFE   [3]  00 EE 00
 (001.250192)  can0  18EEFF50   [8]  88 77 66 55 44 33 22 11
```
Result on Terminal 2:
The cursor does not return. (Which is intended behavior.)

On Terminal 2:
Pressed ctrl-C in order to terminate still running j1939acd.

On Terminal 3 :
Run same j1939acd-command as above, but with '-v' option.

`j1939acd -r 100,80-120 -v -c /tmp/1122334455667788.jacd 1122334455667788`

Result on Terminal 1:
Nothing happens. Expected behavior is the same output as above without '-v' option. 

Result on Terminal 3:
`j1939acd: ready for can0:1122334455667788: Success`

and the cursor returns. Expected behavior the program should continue to run.

olerem, 2020-06-12:
Hi @josemic ,

can you please test it https://github.com/linux-can/can-utils/pull/220

josemic, 2020-06-12:
Tested  olerem / can-utils branch j1939-verbose . It works. j1939acd with -v option does no longer return and provides the following useful output:

```
./j1939acd -r 100,80-120 -v -c /tmp/1122334455667788.jacd 1122334455667788
- ready for can0:1122334455667788
- socket(PF_CAN, SOCK_DGRAM, CAN_J1939);
- setsockopt(, SOL_CAN_J1939, SO_J1939_FILTER, <filter>, 96);
- setsockopt(, SOL_SOCKET, SO_BROADCAST, 1, 4);
- bind(, can0:1122334455667788, 24);
- socket(PF_CAN, SOCK_DGRAM, CAN_J1939);
- setsockopt(, SOL_CAN_J1939, SO_J1939_FILTER, <filter>, 96);
- setsockopt(, SOL_SOCKET, SO_BROADCAST, 1, 4);
- bind(, can0:1122334455667788, 24);
- sendto(, { 0, 0xee, 0, }, 3, 0, -,0ea00, 24);
- request sent, pending for 1250 ms
- bind(, can0:1122334455667788, 24);
- send(, 1234605616436508552, 8, 0);
- claimed 0x50
```
Please merge to linux-can/can-utils branch master.

marckleinebudde, 2020-06-17:
Fixed with https://github.com/linux-can/can-utils/pull/220

closed 2020-06-17 by marckleinebudde