linux-can/can-utils #207: Request for j1939dump and j1939player

josemic, 2020-05-28:
This is a request for discussion for a j1939player, similar to canplayer, which can replay j1939 message.

For now it is assumed that the output of j1939dump is similar to j1939spy. With additionally the option to assemble j1939 TP and ETP messages.
Additionally timestamps are needed, in order to keep the timing during replay.

I'd like to see a j1939player, which can replay messages based upon the J1939-NAME(s), optionally filtered by PGN(s). This would allow to test j1939 application message decoders e.g. of Isobus messages.

marckleinebudde, 2020-05-28:
First we need a way to save complete (E)TP messages in user space. I see two options:

1. Extend the kernel's j1939 socket to receive (E)TP into user space without actively taking part in the communication.
2. Re-assemble (E)TP from a can-dump in user space.

josemic, 2020-05-28:
Perhaps we have a third option (or is that the same as 2.?):
3. Have tool that postprocesses log files and reassembles (E)TP messages
   E.g.:
   cat mydump.log | ETPandTPpostprocessor > mydump_postprocessed.log
   j1939player -NAME 1122334455667788 mydump_postprocessed.log

marckleinebudde, 2020-05-28:
ACK, this is what I meant with 2.

josemic, 2020-05-30:
One difference is that 1. is working for realtime tracing with tools too. E.g. a (to be written) wireshark j1939 decoder. While 2. will never work with realtime trace tools. 
The benefit of 2 is, that it keeps the complexity out of the kernel.

marckleinebudde, 2020-06-02:
If the kernel can re-assemble RAW CAN frames to a j1939 (E)TP, this should also be possible in user space. Maybe it doesn't fit into the programming model of wireshark. On the other hand wireshark can decode TCP, which spans over sevel IP packages....

josemic, 2020-06-30:
I think a j1939player needs to replay besides messages accoding to its name, all messages to global address (DA=0xFF) too.