linux-can/can-utils #219: Generate simple CAN messages

gsus24, 2020-06-08:
Hello
its not an issue, its a starting point. Hope its ok asking here.
 I playing around with CARLA Simulator for autonomous driving research 
[https://github.com/carla-simulator/carla](url)
Its possible to read values for speed, acceleration, throttle, brakepedal, ...
My hope is to put this values in CAN messages.

Any suggestions on doing that are welcome.
Thanks and best regards

marckleinebudde, 2020-06-08:
Hey @gsus24,

what's your use case? Control your real car with the data from the simulation?

There are several project out there that do the de-coding of the CAN buses of real vehicles. Have a look how to map the data values from the simulation to the raw CAN frames. You probably want to re-use some kind of mapping file/database/etc from other projects.

Then look at https://github.com/linux-can/can-utils/blob/master/cansend.c how to open a socket, assemble a CAN frame and send it out. I think there is a python binding for CAN.

gsus24, 2020-06-29:
Hi @marckleinebudde 
i want to get telemetry data for simulation like throttle and brakepedalposition, speed, gear, steering angle... out of CARLA Simulator via CAN-bus and receive and the messages in my program together with Images and drive the car autonomously.
Thanks for the tip with cansend. I used it as workaround with systemcommand for validating the receive.
I take a look for pyhton implementation and find this
[https://github.com/eerimoq/cantools](url)

closed 2020-06-29 by gsus24