linux-can/can-utils #226: App request: limitations of cangw

CanCanRep, 2020-06-30:
Hi and thanks for that great work.
I´m thinking about to use this kernel module in order to implement some kind of a can switch with tracing options inside a testing environment.  I couldn´t find any limitations or measures.

My question are: 
1. What is the maximum limit of Channels ? I´ve approached PEAK Systems and they stated that the socketcan has not limitations of channels. But what about the CANGW? Do you have any experience with that ? E.g. decreasing performance with the number of channels in use ?
2. Do you now an exact value for the delay of the CANGW? Which means the time difference between rx ch0 to tx ch1 for example.

hartkopp, 2020-07-01:
>     1. What is the maximum limit of Channels ? I´ve approached PEAK Systems and they stated that the socketcan has not limitations of channels. But what about the CANGW? Do you have any experience with that ? E.g. decreasing performance with the number of channels in use ?

What do mean with 'channels'?

1. The number of CAN interfaces?
2. The number of routing jobs performed by can-gw?

Anyway both numbers are not limited and designed to scale with max performance. 

>     2. Do you now an exact value for the delay of the CANGW? Which means the time difference between rx ch0 to tx ch1 for example.

The University of Prague did some measurements some time ago which should not have changed from a quality perspective:
http://rtime.felk.cvut.cz/can/benchmark/3.0/
http://canbus.pages.fel.cvut.cz/#can-bus-channels-mutual-latency-testing

Long story short: When the CAN frame is received the NET-RX softirq is run. In  can-gw the frame might be modified and is then directly pushed to the tx queue of the outgoing CAN netdevice. Depending on the tx queue fill state and the speed of the CAN interface the CAN frame is send on the bus.

Due to the possibility that you might have to deal with concurrent outgoing CAN traffic on that interface and the fact that the NET-RX softirq is also handling other network traffic (e.g. ethernet) there is no _fixed_ delay between incoming and outgoing frames (see benchmarks).

CanCanRep, 2020-07-02:
thanks for the fast reply. 

Yes, with channels I meant the maximum of usable HW CAN Interfaces. I´m thinking to apply around 36 Interfaces and do a routing between them without any rules / modifications. E.g. 6 of those [1]. But the maximum of simultaneous active Interfaces is around 10. Do you have experience in that high number of interfaces ? I should strive for a maximum delay of 1ms.

thank you for pointing me to the benchmark tests of the can-gw. I do not completely understand the graphs. But what I got from [2] is that they applied 9VCANs and the maximum delay of frames was <= 0.9ms right with 9VCANs. Applying this to a real HW I have to add some more ns due to HW operations. 
Am I right?

[1]
https://www.peak-system.com/PCAN-USB-X6.438.0.html

[2]
http://rtime.felk.cvut.cz/can/benchmark/3.0/host-2.6.36.2-00015-g4f1bc4e/3.0.4-rt14-00008-gb2052fd/oneatatime/cpu/vcans/results.html

hartkopp, 2020-07-02:
At Volkswagen I set up a system with 20 CAN busses (5x Peak PCIe cards) and 30 vcans in this system:

https://youtu.be/Mu8EfXh3qHw?t=89

We separated ECUs from their CAN busses to perform some modifications. At the end the Intel i7 Low Power (2 CPUs with HT -> 4 CPUs) was routing/modifying/forwarding about 22.000 frames per second - and the system load was about 6-7%.
The car that you can see in the video is driving properly and met all the timing requirements, but I did not make any measurements about delays in this living system. I probably should do that when driving the car the next time ...

CanCanRep, 2020-07-02:
thank you for pointing that out. I think it can be a reliable solution for the strived setup. And it´s somehow comparable to your setup inside a car.  I think I will setup a first PoC. Thanks 

and btw congratulations for your nice job at Volkswagen ;)

closed 2020-12-12 by hartkopp