linux-can/can-utils #233: The timestamps displayed by "candump" are not strictly increasing

Delgan, 2020-08-17:
Hi.

While using the `candump -L can0` command I noticed the following output:

```
(1597294762.404524) can0 0A2CA765#0000000000000000
(1597294762.404562) can0 0A2BA765#0000000000000000
(1597294768.621593) can0 0C4065A7#AF00000058020000
(1597294762.409544) can0 0A25A765#0000000000000000
(1597294762.409661) can0 0A24A765#0000000000000000
```

You can see in the middle the timestamp jumps from `1597294762` to `1597294768` and then go back to `1597294762`.

It's important to note that frames `0A..A765` are sent by us while the `0C4065A7` is a received one. I know each CAN frame can be associated with a timestamp, but I expected `candump` to display the system timestamp and not the CAN one.  Could the timestamp difference be explained by a clock de-synchronization between the system receiving the frames (where `candump` is run) and the system sending them?

Also, I use the `can-utils` from Ubuntu 16.04, it's quite an old release. Maybe this is a bug already fixed on `master`, are you aware of such issue in the past?

hartkopp, 2020-08-17:
Huh! The problem can occur when you have a multicore system and different cores handle the (same) interrupt the CAN controller is connected to. But I've never seen this effect with a 6 seconds hop. o_O
What kind of hardware are you using in your setup?

Delgan, 2020-08-17:
@hartkopp We are using a PCAN-USB (Peak System) connected on a computer running Intel Celeron.

So, can you confirm that the timestamp displayed with `candump -L` should be the one at the time the message is received? This is not related to the system clock of the computer emitting the message?

olerem, 2020-08-17:
There is one more issue on 32bit system. Set:
date -s @2147483647

and run candump with timestamps :)

hartkopp, 2020-08-17:
> @hartkopp We are using a PCAN-USB (Peak System) connected on a computer running Intel Celeron.
> 
> So, can you confirm that the timestamp displayed with `candump -L` should be the one at the time the message is received? 

Yes. The timestamp is generated at frame reception-time inside the network stack on the receiving host system.

How many CPU cores does your Celeron have?

hartkopp, 2020-08-17:
> There is one more issue on 32bit system. Set:
> date -s @2147483647
> 
> and run candump with timestamps :)

```(1597689414.860086)  vcan0  3D2   [8]  78 F9 21 0D C0 58 29 73
 (1597689415.060440)  vcan0  0A2   [6]  41 36 A6 77 13 E5
 (1597689415.260558)  vcan0  05F   [5]  91 89 7B 30 96
 (1597689415.460673)  vcan0  4BA   [8]  CA A4 BD 48 FD F4 3F 09
 (1597689415.660844)  vcan0  224   [8]  1E 20 3B 49 48 23 AD 7F
 (2147483647.090077)  vcan0  450   [8]  68 A8 4D 0D 99 1F E6 30
 (2147483647.290169)  vcan0  79A   [3]  84 50 B0
 (2147483647.490323)  vcan0  5BB   [8]  82 CB D8 23 8D 91 58 43
 (2147483647.690470)  vcan0  7AB   [8]  4E EA 81 36 4D 30 75 7F
 (2147483647.890630)  vcan0  620   [8]  60 15 28 75 80 76 0F 37
 (2147483648.090756)  vcan0  3A4   [1]  16
 (2147483648.290863)  vcan0  3CD   [0] 
 (2147483648.491028)  vcan0  49F   [8]  39 89 1C 64 F0 A8 25 1D
 (2147483648.691213)  vcan0  217   [1]  89
 (2147483648.891396)  vcan0  544   [8]  C8 FB 01 1E 00 73 AE 1C
 (2147483649.091568)  vcan0  40B   [8]  8D 04 07 60 B7 FB 78 0F
 (2147483649.291731)  vcan0  444   [8]  04 2C EE 0E 65 72 C6 72
 (2147483649.491902)  vcan0  76A   [5]  E5 E8 D5 29 0E
 (2147483649.692075)  vcan0  056   [8]  6C 43 F8 20 23 8C E6 66
 (2147483649.892199)  vcan0  5DB   [8]  14 F2 55 4B 7A 82 38 5D
 (2147483650.092442)  vcan0  642   [8]  6A 2B 5E 7A 59 98 34 01
 (2147483650.292632)  vcan0  4EF   [3]  0A 9A B4
 (2147483650.492801)  vcan0  501   [2]  34 2D
```

Oh - this was a 64 bit system :-)

Have to get my Samsung X20 ...

olerem, 2020-08-18:
iMX6:
```
 (1583020909.559575)  can0  49C   [8]  7F 6C F6 12 44 FB 2C 1C                                                                                                                                                                                                                                                                                 
 (1583020909.759742)  can0  1DB   [8]  95 E7 2A 41 57 83 1D 67                                                                                                                                                                                                                                                                                 
 (1583020909.959951)  can0  414   [8]  04 04 36 20 0D 5B B4 2B                                                                                                                                                                                                                                                                                 
 (1583020910.160072)  can0  056   [8]  D8 04 CE 2A 95 86 1B 16                                                                                                                                                                                                                                                                                 
 (1583020910.360017)  can0  617   [1]  53                                                                                                                                                                                                                                                                                                      
 (1583020910.560430)  can0  34B   [8]  91 FD 6C 63 48 5D 43 25                                                                                                                                                                                                                                                                                 
 (2147483647.123279)  can0  76B   [2]  14 3C                                                                                                                                                                                                                                                                                                   
 (2147483647.323607)  can0  292   [8]  E7 69 82 72 2E 1F 12 02                                                                                                                                                                                                                                                                                 
 (2147483647.523745)  can0  475   [7]  72 1A 3F 1E 51 C6 40                                                                                                                                                                                                                                                                                    
 (2147483647.723958)  can0  1C2   [8]  A8 49 5E 0F D6 15 B0 4A                                                                                                                                                                                                                                                                                 
 (2147483647.924134)  can0  137   [8]  E3 70 64 76 8D F9 CA 1A                                                                                                                                                                                                                                                                                 
 (-2147483648.124369)  can0  69B   [8]  23 80 E6 30 B2 FC E0 72                                                                                                                                                                                                                                                                                
 (-2147483648.324481)  can0  56C   [6]  96 1B FD 2D B8 68                                                                                                                                                                                                                                                                                      
 (-2147483648.524730)  can0  42F   [8]  00 C6 27 70 9B 73 54 62                                                                                                                                                                                                                                                                                
 (-2147483648.724759)  can0  12A   [4]  32 65 1A 5A                                                                                                                                                                                                                                                                                            
 (-2147483648.925075)  can0  541   [8]  EB 5A 6C 63 B7 B1 0F 24                                                                                                                                                                                                                                                                                
[... cut ...]                                                                                                
 (-2147483647.925744)  can0  1C4   [2]  DB C9                                                                                                                                                                                                                                                                                                  
 (-2147483646.126094)  can0  76A   [8]  8E 31 CD 59 94 A0 19 78 
```

Delgan, 2020-08-18:
> Yes. The timestamp is generated at frame reception-time inside the network stack on the receiving host system.
> 
> How many CPU cores does your Celeron have?

@hartkopp It's an Intel Celeron N2930, it has 4 cores.

So, the function generating the timestamp seems quite low level, this means the problem is maybe not related to `can-utils`?
The timestamp offset is consistent for each message received. That's weird, I don't understand how the timestamp sources could be different and differ like this.

hartkopp, 2020-08-19:
Yes. Six seconds is a huge gap. The candump tool get's the CAN frame together with the receiving timestamp with the recvmsg() system call, see:
https://github.com/linux-can/can-utils/blob/master/candump.c#L639
And then the data is either provided by a SO_TIMESTAMP or SO_TIMESTAMPING tag:
https://github.com/linux-can/can-utils/blob/master/candump.c#L666

So candump only relies on values provided by Linux syscalls.

Interestingly the PCAN USB adapter can provide hardware timestamps
https://elixir.bootlin.com/linux/v5.8/source/drivers/net/can/usb/peak_usb/pcan_usb.c#L522
but I don't think this causes the outlier of 6 secs ...

Delgan, 2020-08-24:
Thanks for the details, @hartkopp!

Then I'm not sure there's much left to do. Unless anyone has an idea of how to investigate this issue, I don't think I have the required knowledge to understand more deeply the problem. Maybe an update of my system will fix it.

quietscorpion, 2021-02-08:
Hi, @hartkopp, I've stumbled to a similar problem with this, but I use Virtual CAN running on Raspberry Pi (4 CPU cores). The CAN frames would be swapped on a very rare occurrence (once in few millions frames), although my app only has 1 thread. After reading your comment, I tried to run my app using only one core, and the swapped frames problem seems to disappear.
https://github.com/linux-can/can-utils/issues/233#issuecomment-674810149

The problem is, I still want to utilize all 4 CPU cores. Can you give me any advice to do so, while avoiding the swapped frame issues. Thank you.

groengpx, 2023-08-22:
I run candump regurlarly for hours, and I see sometimes a ~10 minute difference between outgoing and ingoing frames. The timestamp from the outgoing frames seem to be correct. It is the incoming part that seems to be off. But not always: sometimes it is entirely correct.

I always run it with 'any' specified as the interface.
Most of the time, two interfaces are connected (can0 and can1).

My can-utils version is 2018.02.0 (from Ubuntu Focal Fossa 20.04 LTS).
My is often close to latest. My last instance of this bug was on 6.4.7 (linux surface kernel).

hartkopp, 2023-08-25:
Please remove the old 2018.02.0 version from Ubuntu (apt remove --purge can-utils) and clone this repository to build and install the latest can-utils tools (just `make && make install`). Not sure if there was a fix in the last 5 years that might have fixed your problem.

marckleinebudde, 2023-10-03:
Any news on this? Feel free to reopen.

closed 2023-10-03 by marckleinebudde