linux-can/can-utils #245: can-j1939 fails on kernel 5.7

zeljkoavramovic, 2020-09-30:
Following can-j1939 kickstart guide does not work for me.

> avra@vm-debian:~/can/can-utils$ sudo modprobe can-j1939
> modprobe: FATAL: Module can-j1939 not found in directory /lib/modules/5.7.0.0.bpo.2-amd64

> avra@vm-debian:~/can/can-utils$ testj1939
> testj1939: socket(j1939): Protocol not supported

I can find can.ko, can-raw.ko and vcan.ko on my system, but nothing similar to can-j1939.ko.

My virtual machine is Debian 10.5 testing with kernel 5.7 from backports:
> avra@vm-debian:~/can/can-utils$ uname -a
> Linux vm-debian 5.7.0.0.bpo.2-amd64 #1 SMP Debian 5.7.10-1-bpo10+1 (2020-07-30) x86_64 GNU/Linux

vcan0 is up and working, with sending and receiving frames without problems. Kayak is also working well over socketcand.

Kickstart guide fails at modprobe can-j1939 so I am stuck. Could you please give me a pointer on what could I try next?

My issue is similar to this one: https://github.com/linux-can/can-utils/issues/197, but I am using newer kernel where this should work. What am I doing wrong?

marckleinebudde, 2020-09-30:
`j1939` is not enabled by default in debian's kernel. So you have to compile these kernel module manually.

zeljkoavramovic, 2020-10-01:
Thank you very much for your help! I see now that my problem is related to kernel module prerequisite.

I have compiled and installed a new kernel following instructions from https://docs.automotivelinux.org/docs/en/master/apis_services/reference/signaling/3-Installation-J1939.html, with an exception that I have used Debian patched kernel instead of stock one. Unfortunately, I was still not able to bring can-j1939 up:

> avra@vm-debian:~/can/can-utils$ sudo modprobe can-j1939
> modprobe: FATAL: Module can-j1939 not found in directory /lib/modules/5.7.10

vcan0 works well as it did with previous kernel. 

Here is additional info:
> avra@vm-debian:~/can/can-utils$ uname -a
> Linux vm-debian 5.7.10 #1 SMP Wed Sep 30 15:17:06 CEST 2020 x86_64 GNU/Linux

I am confused since I can find can-j1939.ko in /lib/modules/5.7.10/kernel/net/can/j1939 directory. I have also found that can-j1939 text is mentioned in these modules files:
> /lib/modules/5.7.10/modules.dep
> /lib/modules/5.7.10/modules.order
> /lib/modules/5.7.10/modules.dep.bin

Do you have any idea what could I try next?

**UPDATE:** It seams that I was able to bring can-j1939 up after all using insmod. I will see what I need to do to fix modprobe.

If I find that can-j1939 is really working, would you be interested that I make can-j1939-install-kernel-module.md?

Thank you for your support and nice work!

marckleinebudde, 2020-10-01:
> UPDATE: It seams that I was able to bring can-j1939 up after all using insmod. I will see what I need to do to fix modprobe.

try `sudo depmod -av`

marckleinebudde, 2020-10-01:
> If I find that can-j1939 is really working, would you be interested that I make can-j1939-install-kernel-module.md?

Can you start writing it, as you just managed to build one. Create a pull request and I'll review it.

zeljkoavramovic, 2020-10-01:
I will after testing that can-j1939 really works. Thanks!

zeljkoavramovic, 2020-10-05:
can-j1939 works in my vm. Thanks!

As promised I have written can-j1939-install-kernel-module.md file and created a pull request:
https://github.com/zeljkoavramovic/can-utils/pull/1

However I do not see it in official list yet:
https://github.com/linux-can/can-utils/pulls

Please review and tell me if I need to do anything else.

zeljkoavramovic, 2020-10-06:
Another pull request https://github.com/linux-can/can-utils/pull/247

closed 2020-10-06 by zeljkoavramovic