linux-can/can-utils #257: canbusload not including its own device?

helo9, 2020-10-31:
First of all, thanks a lot for the awesome software!

I am playing around with canbusload and a socketcan device and was wondering if the load generated by the device itself is not included within the results shown by canbusload.

marckleinebudde, 2020-11-02:
Hey @helo9,

with normal configuration, the generated packets are included in the bus load calculation.

With `cansequence -pv can1` I'm only sending CAN frames and `canbusload` shows:

```
canbusload 2020-11-02 11:46:34 (worst case bitstuffing)
 can1@500000  3122  202930  24976  40% |XXXXXXXX............|
```

helo9, 2020-11-02:
Thanks @marckleinebudde, was able to reproduce now. Not sure what was wrong back then. Thanks!

I'll close the issue ;)

closed 2020-11-02 by helo9