linux-can/can-utils #263: Why candump capture singals(SIGTERM,SIGUP,SIGINT)

oak-tree, 2020-12-03:
Hey,

According to https://github.com/linux-can/can-utils/blob/c3d9cc2027d261dbde8b5fe6d82de87cd0ed5596/candump.c#L252
candump captures and cancel the following signals: `SIGTERM,SIGUP,SIGINT`
any specials reason it does so?

Thanks

marckleinebudde, 2020-12-03:
Hey @oak-tree,

what do you mean by "cancel" a signal? These calls install a signal handler function called `sigterm`

https://github.com/linux-can/can-utils/blob/c3d9cc2027d261dbde8b5fe6d82de87cd0ed5596/candump.c#L163-L166

which is used to shut down the `while` loop in `main()`.

https://github.com/linux-can/can-utils/blob/c3d9cc2027d261dbde8b5fe6d82de87cd0ed5596/candump.c#L616

oak-tree, 2020-12-03:
@marckleinebudde 
I Do not know how I missed that, the signals are there to cancel the infinity loop when needed.
Cool thanks

closed 2020-12-03 by oak-tree