linux-can/can-utils #272: valgrind said  socketcall.bind(my_addr.sa_data) points to uninitialised byte(s)

ju-emb, 2020-12-18:
Not sure that's an issue, my fault or some valgrind bug.

The program works fine on a beaglebone black with debian 10 and socket can from master.
Running it on valgrind pops the error

==3764== Syscall param socketcall.bind(my_addr.sa_data) points to uninitialised byte(s)
==3764==    at 0x49F51C8: bind (syscall-template.S:78)
==3764==    by 0x10887D: open_port (bb_CANTest.c:35)
==3764==    by 0x108AAD: main (bb_CANTest.c:92)
==3764==  Address 0xbde0f542 is on thread 1's stack
==3764==  in frame #1, created by open_port (bb_CANTest.c:13)
==3764==  Uninitialised value was created by a stack allocation
==3764==    at 0x1087FA: open_port (bb_CANTest.c:13)

Any advice?
Juergen

my code
```c
#include <stdio.h>
#include <string.h>
#include <unistd.h>
#include <fcntl.h>
#include <sys/ioctl.h>
#include <net/if.h>
#include <linux/can.h>
#include <linux/can/raw.h>

int soc;
int read_can_port;

int open_port(const char *port) {
	struct ifreq ifr;
	struct sockaddr_can addr;

	/* open socket */
	soc = socket(PF_CAN, SOCK_RAW, CAN_RAW);
	if (soc < 0) {
		return (-1);
	}

	addr.can_family = AF_CAN;
	strcpy(ifr.ifr_name, port);

	if (ioctl(soc, SIOCGIFINDEX, &ifr) < 0) {

		return (-1);
	}

	addr.can_ifindex = ifr.ifr_ifindex;

	fcntl(soc, F_SETFL, O_NONBLOCK);

	if (bind(soc, (struct sockaddr*) &addr, sizeof(addr)) < 0) {

		return (-1);
	}

	return 0;
}

int send_port(struct can_frame *frame) {
	int retval;
	retval = write(soc, frame, sizeof(struct can_frame));
	if (retval != sizeof(struct can_frame)) {
		return (-1);
	} else {
		return (0);
	}
}

/* this is just an example, run in a thread */
void read_port() {
	struct can_frame frame_rd;
	int recvbytes = 0;

	read_can_port = 1;
	while (read_can_port) {
		struct timeval timeout = { 1, 0 };
		fd_set readSet;
		FD_ZERO(&readSet);
		FD_SET(soc, &readSet);

		if (select((soc + 1), &readSet, NULL, NULL, &timeout) >= 0) {
			if (!read_can_port) {
				break;
			}
			if (FD_ISSET(soc, &readSet)) {
				recvbytes = read(soc, &frame_rd, sizeof(struct can_frame));
				if (recvbytes) {
					printf("ID 0x%X dlc = %d, data = ", frame_rd.can_id, frame_rd.can_dlc);
					for (int i = 0; i < frame_rd.can_dlc; i++) {
						printf("0x%X ", frame_rd.data[i]);
					}
					printf("\n");

				}
			}
		}

	}

}

int close_port() {
	close(soc);
	return 0;
}

int main(void) {
	open_port("can0");
	read_port();
	close_port();
	return 0;
}

```

hartkopp, 2020-12-18:
Hi Juergen,

struct sockaddr_can contains much more space than needed in the code example:

```/**
 * struct sockaddr_can - the sockaddr structure for CAN sockets
 * @can_family:  address family number AF_CAN.
 * @can_ifindex: CAN network interface index.
 * @can_addr:    protocol specific address information
 */
struct sockaddr_can {
        __kernel_sa_family_t can_family;
        int         can_ifindex;
        union {
                /* transport protocol class address information (e.g. ISOTP) */
                struct { canid_t rx_id, tx_id; } tp;

                /* J1939 address information */
                struct {
                        /* 8 byte name when using dynamic addressing */
                        __u64 name;

                        /* pgn:
                         * 8 bit: PS in PDU2 case, else 0
                         * 8 bit: PF
                         * 1 bit: DP
                         * 1 bit: reserved
                         */
                        __u32 pgn;

                        /* 1 byte address */
                        __u8 addr;
                } j1939;

                /* reserved for future CAN protocols address information */
        } can_addr;
};
```

But you only used the first two elements and left the rest uninitialized:

```
        __kernel_sa_family_t can_family;
        int         can_ifindex;
```

From that standpoint valgrind is right :-)

marckleinebudde, 2020-12-18:
> int open_port(const char *port) {
> 	struct ifreq ifr;
> 	struct sockaddr_can addr;
Use C99 initializers here:

```
struct sockaddr_can addr = {
	.can_familiy = AF_CAN,
};
```
to initialize the rest with zero.

hartkopp, 2020-12-18:
Not bad! (wie der Schwabe sagen würde)

ju-emb, 2020-12-18:
Thanks to both of you.
Works as a charm. 

If you do it right it just works. (sagt der Deutsche in Peru)

closed 2020-12-18 by ju-emb