linux-can/can-utils #273: Not an issue - cangw drop

vsboost, 2021-01-03:
Newbie here, could i please get some assistance with simply dropping frame  "can1  3E7  [4] FF FF 7C 34" -- simply don't want it to pass through the gw but don't understand the correct command line.

Cheers

hartkopp, 2021-01-03:
Puh - so lucky that it's not an issue ;-)

The filter (which is analogue to the candump filters) need to have a `~` instead of `:` to negate the filter:

`cangw -A -s can0 -d can1 -f 3E7~C00007FF`

or if you want to test it on virtual CAN interfaces you should switch on the '-e' option:

`cangw -A -s vcan0 -d vcan1 -f 3E7~C00007FF -e`

vsboost, 2021-01-04:
Awesome, thanks so much for that - working well.

vsboost, 2021-01-04:
Hi Oliver,

I thought i had this sorted but apparently not, seems when i use this rule it blocks all messages from id 3E7, instead i just need it block one message (FF FF 7C 34) and let the rest pass through the gateway, how would i go about implementing this?

Thanks again for your help

hartkopp, 2021-01-04:
Oh, that makes it more tricky.

Once you have enabled the notch filter from above which eliminates 3E7 you can define *another* can-gw rule which handles only 3E7 and can modify the content with AND/OR/XOR/SET bit operations if this helps for your use case.

When you can't fulfill your requirements with one or more of these bit operations and *really* need to check for the exact FF FF 7C 34 content, I only see the CAN_RAW socket as solution right now. But if you use the CAN_RAW socket for routing, the timings in userspace are usually worse in their jitter. Additionally you should still stay for the can-gw notch filter from above and only filter for the 3E7:C00007FF to get only this CAN ID via CAN_RAW to reduce the load of CAN traffic which is routed via user space.

vsboost, 2021-01-04:
Thanks for the tips, i will give this a go.

closed 2021-01-04 by vsboost