linux-can/can-utils #279: Candump no data?

johndoster56, 2021-03-04:
I'm working with Ubuntu 20.04, arduino R3, seeduino V2 can bus shield. I've 'successfully' configured the device but cannot see data transmission. Can confirm data link operation and integrity/ activity with oscilloscope. I have successfully seen data under send/ receive terminals via vcan, but after attempting to implement on GMLAN in seeing nothing. Please advise as this has become a frustrating venture. 
[image]
[image]
[image]

johndoster56, 2021-03-04:
After reviewing issue 170 I believe this applies in the arduino coding and has something to do with it
[image]

hartkopp, 2021-03-05:
So this issue can be closed?

johndoster56, 2021-03-05:
> So this issue can be closed?

I wouldn't say so. Not yet,  I read about it some last night,  will post more when time allows

closed 2021-04-17 by hartkopp