linux-can/can-utils #280: canplayer from logfile step by step or 1 frame at a time

nico0481, 2021-03-09:
Hello,

Is it possible to send frames from file but rather "step-by-step" (frame by frame) using a key or using a gap between frames. I've tried with -g (X ms), but it seems to send lot of frames every X ms, not frame by frame.

Thank you

Best regards

hartkopp, 2021-04-17:
Hi,
I added a patch to my private repo here: https://github.com/hartkopp/can-utils/commit/aaa7fd215f4ecde45d2047d8ebfaed3c255d66d7
So you can just clone it and do some tests based on that latest commit.
As the commit message says, it doesn't win a design price ;-)
But it was easier than writing an excuse why your use-case is uncommon ...
Would this approach be ok for you?

hartkopp, 2021-05-28:
@nico0481 Ping?

nico0481, 2021-06-04:
Hello,

It works great!! Many thanks
Sorry for the late feedback.

Best regards

closed 2021-06-04 by nico0481