linux-can/can-utils #281: CAN dump loopback off not working

alessio0208, 2021-03-09:
Hello, I recently had to reinstall Raspbian on my Rasperry Pi. As a consequence, my PiCAN DUO now does not collect anymore data with candump. I am pretty sure that it is a SW configuration problem rather than HW, because prior to the OS reinstallation, I successfully collected data on multiple vehicles.

Right now, candump with the loopback on works fine. However, when the loopback is off and the PiCAN is connected to the car, it keeps waiting and it does not receive anything. I also tried to stimulate some reply from the CAN bus, sending the cansend can0 7DF#0201050000000000. Still nothing happend.

The output of   **ip -det link show can0** is:

can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
	    link/can  promiscuity 0 minmtu 0 maxmtu 0 
	    can state ERROR-ACTIVE restart-ms 0 
		  bitrate 500000 sample-point 0.875 
		  tq 125 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
		  mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
		  clock 8000000numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 

What could be the cause *can state ERROR-ACTIVE* and what does it imply? Could you please help me?
Thanks in advance

marckleinebudde, 2021-03-09:
Hey alessio0208,

`ERROR-ACTIVE` means the CAN controller is in good shape, has not detected a huge amount of errors. See https://www.kvaser.com/about-can/the-can-protocol/can-error-handling/#:~:text=A%20node%20starts%20out%20in,Flags%20when%20it%20detects%20errors.

What is the version of the old and new kernel?
Can you send me the output of `cat /proc/interrupts`.

regards,
Marc

alessio0208, 2021-03-09:
Hi, thank you very much for your quick reply!

The current kernel is 5.4.83. Not sure about the previous one, most likely 4.19 or 5.4.51.

Output of cat /proc/interrupts:

           CPU0       CPU1       CPU2       CPU3       
 17:       8573          0          0          0  ARMCTRL-level   1 Edge      3f00b880.mailbox
 18:        903          0          0          0  ARMCTRL-level   2 Edge      VCHIQ doorbell
 40:          0          0          0          0  ARMCTRL-level  48 Edge      bcm2708_fb DMA
 42:        690          0          0          0  ARMCTRL-level  50 Edge      DMA IRQ
 44:       7569          0          0          0  ARMCTRL-level  52 Edge      DMA IRQ
 45:          0          0          0          0  ARMCTRL-level  53 Edge      DMA IRQ
 48:          0          0          0          0  ARMCTRL-level  56 Edge      DMA IRQ
 56:      91495          0          0          0  ARMCTRL-level  64 Edge      dwc_otg, dwc_otg_pcd, dwc_otg_hcd:usb1
 78:          0          0          0          0  ARMCTRL-level  86 Edge      3f204000.spi
 80:       1827          0          0          0  ARMCTRL-level  88 Edge      mmc0
 81:       7542          0          0          0  ARMCTRL-level  89 Edge      uart-pl011
 86:     349927          0          0          0  ARMCTRL-level  94 Edge      mmc1
161:          0          0          0          0  bcm2836-timer   0 Edge      arch_timer
162:      22711      42108      18095      15977  bcm2836-timer   1 Edge      arch_timer
165:          0          0          0          0  bcm2836-pmu   9 Edge      arm-pmu
166:          2          0          0          0  lan78xx-irqs  17 Edge      usb-001:004:01
167:        101          0          0          0  pinctrl-bcm2835  25 Level     spi0.0
FIQ:              usb_fiq
IPI0:          0          0          0          0  CPU wakeup interrupts
IPI1:          0          0          0          0  Timer broadcast interrupts
IPI2:      14764      60344      57297      25518  Rescheduling interrupts
IPI3:        414       1001       1116       2147  Function call interrupts
IPI4:          0          0          0          0  CPU stop interrupts
IPI5:       4952      13752       3331       2575  IRQ work interrupts
IPI6:          0          0          0          0  completion interrupts
Err:          0

alessio0208, 2021-03-10:
It was 4.19. I reinstalled it with the hope of solving the problem, but it still doesn't work. 

I 

> This means there have been 101 interrupts from the mcp2515 to the CPU:

>          CPU0       CPU1       CPU2       CPU3  
>167:        101          0          0          0  pinctrl-bcm2835  25 Level     spi0.0
>Reboot, then the interrupts should be 0, Retest without loopback mode, make sure you have proper termination, at least two >CAN controllers on the bus. If the interrupts increase, then at least the mcp2515 reacts to the CAN frames.

I tried and the interrupts counter is still 0  after a while. 

Further information: I connected to the CAN bus hrough the OBD-II port with an OBD cable (linked to PiCAN through DB9). I tested the same cable with a CL2000 from csselectronics and I manage to collect the data, so we can exclude it as possible source of the problem. 
I also checked that the terminations of the wires for CAN low, CAN high (and ground) are still correctly attached to both PiCAN and DB9 port. 

Thanks again for the support!

marckleinebudde, 2021-03-10:
Can you connect both CAN interfaces of the PiCAN DUO and send messages from `can0` to `can1`?

marckleinebudde, 2021-03-10:
This means both CANs are working. So the next step is to check the cabling when connecting to your car.

alessio0208, 2021-03-10:
So, I managed to use a multimeter (fortunately, I am at the University) to check that there are no problems with the wires from the PiCAN to the DB9 port and from the DB9 port to the OBD port. All the pins seem to be correctly connected. I am going now to test on the car with can1 and let you know

alessio0208, 2021-03-10:
It works. So I think there is some loose connection with can0. I don't know what I didn't do this test at the beginning. Nonetheless, I learnt a lot, so thank you very much!

closed 2021-03-10 by marckleinebudde