linux-can/can-utils #282: J1939 multiple connections AND how to access registered address in kernel

busenitzb, 2021-03-19:
First of all, I appreciate the work that has been done to integrate the j1939 protocol into the Linux kernel. This will be especially helpful for several tasks I have been assigned.

1. My first question may mostly pertain to my limited understanding of sockets. I am developing an application where I hope to develop a single j1939 "ecu" (with j1939 socketcan) that communicates with several other existing j1939 ecu's on the bus. The thought is that my ecu application will take care of the address claiming bits and then perform peer-to-peer communication with several other ecu's on the bus. I have completed the address claiming portion of the application but cannot determine the correct way to establish multiple peer-to-peer connections with other ecu's. These peer-to-peer connections will have to be enabled / disabled at the users discretion throughout the lifetime of the application. My question how / what is the best way to accomplish this?
- Is it possible to enable / disable different reception filters (SO_J1939_FILTER) using setsockopt() after calling bind()? If so, do I accomplish this by re-creating the j1939_filter array every time I need to enable / disable reception of another ecu on the bus? This j1939_filter array would need to be resized at runtime from time to time as the application attempts to communicate with different ecu's. I have tried std::vector as I am using c++ but setsockopt() does not work as expected using the c++ vector class.
- So, does it make the most sense to open / close a socket for each ecu I want to bind() and connect() to on the bus?

2. Is there a way to access what SA / 64-bit names are "registered" in the kernel? I am managing a "secondary" list of registered address / 64-bit names in my application similar to j1939acd. This seems redundant as this information should already be contained in the kernel.

Thank you.

olerem, 2021-03-20:
> First of all, I appreciate the work that has been done to integrate the j1939 protocol into the Linux kernel.

you are welcome :)

> This will be especially helpful for several tasks I have been assigned.
>
>     1. My first question may mostly pertain to my limited understanding of sockets. I am developing an application where I [...]

There are different ways you can implement what you wont:
- create a new socket for each connection (bind(), connect()) and use close() after job is done. Each ECU/application can have many sockets for the same address. Even more, you can assign multiple applications to the same address. In case of bind(), connect() you will be able to use send()/recv() calls
- if you like to work with one socket, then you should avoid to use connect() and use   recvfrom()/sendto() instead. This will probably make your application more complicated.

> * Is it possible to enable / disable different reception filters (SO_J1939_FILTER) using setsockopt() after calling bind()? If so, do I accomplish this by re-creating the j1939_filter array every [...]

You can replace filters, but it seems to make no real sense. I would suggest you to use multiple sockets. And let kernel do everything needed for you.

> 
>     * So, does it make the most sense to open / close a socket for each ecu I want to bind() and connect() to on the bus?

Yes, open socket for new connection and close it if it is not needed.

>     1. Is there a way to access what SA / 64-bit names are "registered" in the kernel? I am managing a "secondary" list of registered address / 64-bit names in my application similar to j1939acd. This seems redundant as this information should already be contained in the kernel.

Yes, it is redundant information but export interface is currently not implemented. Patches are welcome :)

busenitzb, 2021-03-22:
@olerem , thank you for the information. I appears that using multiple sockets is the more robust design pattern for my application.

closed 2021-03-22 by busenitzb

olerem, 2021-03-22:
@busenitzb , it will be good if you can provide some feedback :)
Some thing like:
- is it working for you?
- did it helped you to reduce some existing code?
- make it more stable/secure?
- some examples?

It is just kind of demotivating to not see results of my work

busenitzb, 2021-03-22:
> is it working for you?

Yes, I have much of the network management portion completed. This was more of a c++ re-write of j1939acd (which was VERY useful).

> did it helped you to reduce some existing code?

There was no existing code base for my application so having j1939 protocol already in the kernel has reduced my development effort.

> make it more stable/secure?

Again, there was no existing code base; however, I am assuming that offloading certain tasks relating to j1939 protocol to the kernel will definitely have a positive affect on performance and stability.

> some examples

In the end, I am developing a J1939-capable ROS package. Currently, there is a socketcan_bridge ROS package but it does not have J1939 capabilities. My objective is to establish a ROS-style J1939 ecu that can create multiple peer-to-peer connections to other ecu's on a bus. In addition, it would be able to handle other J1939 duties such as TP. As you can imagine, this will be a somewhat large application maybe I can share the ROS package when completed.

I am sure I will have more questions moving forward. Is it suitable to open more issues as I have more questions or is there another forum that is more appropriate?

olerem, 2021-03-23:
> I am sure I will have more questions moving forward. Is it suitable to open more issues as I have more questions or is there another forum that is more appropriate?

sure!

marckleinebudde, 2021-03-24:
Hey @busenitzb, no forum but we also have the linux-can mailinglist: http://vger.kernel.org/vger-lists.html#linux-can