linux-can/can-utils #284: J1939 with Raspberry Pi v3 + ubuntu 20.10 server arm64

ManishKakra, 2021-04-14:
Hello All,
How do I add J1939 support in ubuntu 20.10 prebuilt server arm64 image that I have on Raspberry Pi V3 ?

For me socket(PF_CAN, SOCK_RAW, CAN_RAW) works but socket(PF_CAN, SOCK_DGRAM, CAN_J1939) does not work.
Get error: socket(j1939): Protocol not supported

When I do modeprobe it gives below error.
$ sudo modprobe can-j1939
modprobe: FATAL: Module can-j1939 not found in directory /lib/modules/5.8.0-1006-raspi

Please help me on how to add J1939 support ? Thank you!!

olerem, 2021-04-14:
Here what I have on my ubuntu desktop:
dpkg -S /lib/modules/5.8.0-48-generic/kernel/net/can/j1939/can-j1939.ko
linux-modules-5.8.0-48-generic: /lib/modules/5.8.0-48-generic/kernel/net/can/j1939/can-j1939.ko

If it is not included by default, please report a bug to your distribution. And as long as you wait until it will be fixed, you can build your own package with this module. Search for:
"ubuntu how to build kernel module" or "ubuntu build kernel"

ManishKakra, 2021-04-14:
Thank you Olerem for your response. I am using Ubuntu Server 20.10 for raspberry pi 3 from https://ubuntu.com/download/raspberry-pi
I also tried "Raspberry Pi OS with desktop and recommended software" for raspberry pi 3 from
https://www.raspberrypi.org/software/operating-systems/#raspberry-pi-os-32-bit

both does not include the can-j1939.ko by default. Looks like only way to include j1939 stack is to rebuild the kernel.
Since I am new on this..and I didn't find the exact match for my scenario.
It will be very helpful if you can please point me to any specific link to do this for any of above two images that I am using?
I am fine if there is other distribution / image that works on raspberry pi 3.

Thank you in advance.

Manish K

marckleinebudde, 2021-04-14:
https://github.com/linux-can/can-utils/blob/master/can-j1939-install-kernel-module.md

ManishKakra, 2021-04-16:
Thank you so much Marc.. this is helpful but I still have below queries in order to move forward..

1. Compiling kernel from source  can be done only on Raspberry Pi 3 itself or should be on separate host machine  ?

	I have "Raspberry Pi OS with desktop"  image on Pi 3 and when I tried these steps I am stuck at 
	" copy and use our old configuration to run menuconfig
	avra@vm-debian:/usr/src/linux-source-5.8$ sudo cp /boot/config-$(uname -r) .config 
	
	As it is not able to find the config file.. This is what I have boot directory
	pi@raspberrypi:/boot $ ls -a
	 .                          bcm2711-rpi-cm4.dtb   kernel7l.img
	 ..                         bootcode.bin          kernel8.img
	 bcm2708-rpi-b.dtb          cmdline.txt           kernel.img
	 bcm2708-rpi-b-plus.dtb     config.txt            LICENCE.broadcom
	 bcm2708-rpi-b-rev1.dtb     COPYING.linux         overlays
	 bcm2708-rpi-cm.dtb         fixup4cd.dat          start4cd.elf
	 bcm2708-rpi-zero.dtb       fixup4.dat            start4db.elf
	 bcm2708-rpi-zero-w.dtb     fixup4db.dat          start4.elf
	 bcm2709-rpi-2-b.dtb        fixup4x.dat           start4x.elf
	 bcm2710-rpi-2-b.dtb        fixup_cd.dat          start_cd.elf
	 bcm2710-rpi-3-b.dtb        fixup.dat             start_db.elf
	 bcm2710-rpi-3-b-plus.dtb   fixup_db.dat          start.elf
	 bcm2710-rpi-cm3.dtb        fixup_x.dat           start_x.elf
	 bcm2711-rpi-400.dtb        issue.txt            'System Volume Information'
	 bcm2711-rpi-4-b.dtb        kernel7.img
	pi@raspberrypi:/boot $
	
	What I am doing wrong this this ? Is there any different step for this use case ?

	2. Another thing that I am trying it have debian (debian-10.9.0-amd64-netinst.iso) on virtual box and following these steps. Kernel build process with j1939 is going on right now..if that is successful how I can use the image/module on my raspberry pi 3 ?
	
	What output I will get as an output after "sudo make -j $(nproc)" is successful ?
	And will I be able to take that from my host virtual machine to Raspberry Pi ?
	Or will this be a whole image all together that I can flash in micro sd card for Raspberry Pi 3?
	Sorry, if questions are very basics but your direction will surely help.
	
	Thank you in advance.
	
Manish

zeljkoavramovic, 2021-04-21:
> 1. Compiling kernel from source  can be done only on Raspberry Pi 3 itself or should be on separate host machine  ?

If you want to compile kernel on Pi3 then make sure you have enough RAM, bigger swap file then default, and enough time. Search Pi forums for exact amount. You can also cross compile from your main PC, but that is not for a newbie.

> 1. I have "Raspberry Pi OS with desktop"  image on Pi 3 and when I tried these steps I am stuck at
>    " copy and use our old configuration to run menuconfig
>    avra@vm-debian:/usr/src/linux-source-5.8$ sudo cp /boot/config-$(uname -r) .config

Raspberry Pi foundation uses Debian as a base for their Raspberry Pi OS, but since they made a lot of changes, it is normal that things differ. Since uname -r on my machine gave 5.8.10 as a result at the time of writing, that command copies /boot/config-5.8.10 to current directory (/usr/src/linux-source-5.8) as file named .config. You need to figure out kernel file name and adapt cp command accordingly. Search Pi wiki and forum for the info. I do plan to update tutorial for Raspberry Pi and PocketBeagle, but that will not be any time soon.

> 1. Another thing that I am trying it have debian (debian-10.9.0-amd64-netinst.iso) on virtual box and following these steps. Kernel build process with j1939 is going on right now..if that is successful how I can use the image/module on my raspberry pi 3 ?

Many people have followed the steps and succeeded, so you should too. However that will only build kernel for that VM and not for Pi (which has a different CPU). If you want to follow that path you will need to learn a quite lot about cross compilation. As already said, not for a newbie, but decision is yours.

> 1. And will I be able to take that from my host virtual machine to Raspberry Pi ?

Tutorial does not cover cross compilation, so following tutorial steps will not allow that.

>    Or will this be a whole image all together that I can flash in micro sd card for Raspberry Pi 3?

My tutorial does not allow you to build an image for Pi from VM.

ManishKakra, 2021-04-21:
Thanks you zeljkoavramovic for the clear answer.
I am able to add the J1939 and compile the kernel for Debian virtual machine successfully using steps in above link and I am also able to test some of the things.
For cross compilation for Pi I will look into the other forums keeping your direction in mind.
Thank you.

closed 2021-04-21 by ManishKakra