linux-can/can-utils #287: Is it possible to use TCP/IP socket from ELM327 clones?

maxdd, 2021-04-17:
Hi,
as per the title would it be possible to use the ELM327 clones together with these tools?
As far as i can see the ELM327 is providing a WIFI network and via Telnet to port  35000 it is possible to send AT commands as well as OBD PIDs request (maybe less relevant)

https://www.sparkfun.com/datasheets/Widgets/ELM327_AT_Commands.pdf
https://learn.sparkfun.com/tutorials/obd-ii-uart-hookup-guide/obd-commands

as far as i can see using the following guide it should be possible to sniff the CAN

https://www.csmagics.com/single-post/2013/07/24/this-is-the-title-of-your-first-image-post

Regards,

hartkopp, 2021-04-17:
Here is a project that implements the ELM327 raw protocol and make it a SocketCAN compatible device:
https://github.com/norly/elmcan
To interconnect this driver with WIFI you would need some 'netcat' script that redirects the traffic to a tty or pty, so that the above driver can access the content.

Alternatively you might build some program which talks to the ELM327 over WIFI and route the CAN frames to a virtual CAN interface in which you can use the can-utils then.

Regards,
Oliver

maxdd, 2021-04-17:
So as far as i understood i should compile the elmcan and use something along this 

`nc 192.168.0.yy 35000 </dev/ttyCAN0 >/dev/ttyCAN0`

and then

```
sudo ldattach \
       --debug \
       --speed 38400 \
       --eightbits \
       --noparity \
       --onestopbit \
       --iflag -ICRNL,INLCR,-IXOFF \
       29 \
       /dev/ttyCAN0
```

at this point in a terminal what shall i type?

`cansniffer any`

is it ok?

leao-nardo, 2021-06-13:
> So as far as i understood i should compile the elmcan and use something along this
> 
> `sudo socat pty,link=/dev/ttyCAN0,raw tcp:192.168.0.10:35000&`
> 
> and then
> 
> ```
> sudo ldattach \
>        --debug \
>        --speed 38400 \
>        --eightbits \
>        --noparity \
>        --onestopbit \
>        --iflag -ICRNL,INLCR,-IXOFF \
>        29 \
>        /dev/ttyCAN0
> ```
> 
> at this point in a terminal what shall i type?
> 
> `./cansniffer any`
> 
> or looking at the code (i'm no expert)
> 
> `./cansniffer elmcan`
> 
> is it ok?

hows it going ? did you tested it ? did it worked ?

yegorich, 2021-06-13:
If ELM327 firmware can understand SLCAN protocol, you could use the same commands as we do for our NetCAN Plus devices: 

1. `sudo sh -c ”socat pty,link=/dev/netcan0,raw tcp:192.168.1.5:2001 &”`
2. `slcand -c -o -s8 /dev/netcan0` - this command creates `slcan0` CAN device
3. `ip link set up slcan0`

Now you can use `slcan0` device with `candump`, `cansend` etc. like you would do with `can0` device.

For more details, see our [manual](https://www.vscom.de/download/multiio/others/info/VSCAN_Manual.pdf#page=26)

fabHO1, 2023-01-24:
Hello everyone I am working with a raspberry pi 3+ and a PICANDuo I am working with raspbian, I followed the steps in the user manual of the pican to make it work and then I tried to run a sample python code to receive data from the OBDII port but when I run it after a few seconds it sends me some errors including OSError: [Errno 105] No buffer space available even before running the code I modified the socketcan native to only socketcan since the first is no longer for version 4. 0 of python-can and also indicates me some file folders in which there is a file socketcan. py which indicates me some lines this last one do you think it is an error of the socketcan file or is it normal I hope you can help me thanks.

This is the python example I am running from my terminal.

#!/usr/bin/python3
#
## obdii_logger.py
# 
# This python3 program sends out OBDII request then logs the reply to the sd card.
# For use with PiCAN boards on the Raspberry Pi
# http://skpang.co.uk/catalog/pican2-canbus-board-for-raspberry-pi-2-p-1475.html
#
# Make sure Python-CAN is installed first http://skpang.co.uk/blog/archives/1220
#
#  24-08-16 SK Pang
#

import RPi.GPIO as GPIO
import can
import time
import os
import queue
from threading import Thread

led = 22
GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(led,GPIO.OUT)
GPIO.output(led,True)

# For a list of PIDs visit https://en.wikipedia.org/wiki/OBD-II_PIDs
ENGINE_COOLANT_TEMP = 0x05
ENGINE_RPM          = 0x0C
VEHICLE_SPEED       = 0x0D
MAF_SENSOR          = 0x10
O2_VOLTAGE          = 0x14
THROTTLE            = 0x11

PID_REQUEST         = 0x7DF
PID_REPLY           = 0x7E8

outfile = open('log.txt','w')

print('\n\rCAN Rx test')
print('Bring up CAN0....')

# Bring up can0 interface at 500kbps
os.system("sudo /sbin/ip link set can0 up type can bitrate 500000")
time.sleep(0.1)	
print('Ready')

try:
	bus = can.interface.Bus(channel='can0', bustype='socketcan')
except OSError:
	print('Cannot find PiCAN board.')
	GPIO.output(led,False)
	exit()

def can_rx_task():	# Receive thread
	while True:
		message = bus.recv()
		if message.arbitration_id == PID_REPLY:
			q.put(message)			# Put message into queue

def can_tx_task():	# Transmit thread
	while True:

		GPIO.output(led,True)
		# Sent a Engine coolant temperature request
		msg = can.Message(arbitration_id=PID_REQUEST,data=[0x02,0x01,ENGINE_COOLANT_TEMP,0x00,0x00,0x00,0x00,0x00], is_extended_id=False)
		bus.send(msg)
		time.sleep(0.05)

		# Sent a Engine RPM request
		msg = can.Message(arbitration_id=PID_REQUEST,data=[0x02,0x01,ENGINE_RPM,0x00,0x00,0x00,0x00,0x00], is_extended_id=False)
		bus.send(msg)
		time.sleep(0.05)

		# Sent a Vehicle speed  request
		msg = can.Message(arbitration_id=PID_REQUEST,data=[0x02,0x01,VEHICLE_SPEED,0x00,0x00,0x00,0x00,0x00],is_extended_id=False)
		bus.send(msg)
		time.sleep(0.05)		

		# Sent a Throttle position request
		msg = can.Message(arbitration_id=PID_REQUEST,data=[0x02,0x01,THROTTLE,0x00,0x00,0x00,0x00,0x00],extended_id=False)
		bus.send(msg)
		time.sleep(0.05)
		
		GPIO.output(led,False)
		time.sleep(0.1)
						
						
q = queue.Queue()
rx = Thread(target = can_rx_task)  
rx.start()
tx = Thread(target = can_tx_task)
tx.start()

temperature = 0
rpm = 0
speed = 0
throttle = 0
c = ''
count = 0

# Main loop
try:
	while True:
		for i in range(4):
			while(q.empty() == True):	# Wait until there is a message
				pass
			message = q.get()

			c = '{0:f},{1:d},'.format(message.timestamp,count)
			if message.arbitration_id == PID_REPLY and message.data[2] == ENGINE_COOLANT_TEMP:
				temperature = message.data[3] - 40			#Convert data into temperature in degree C

			if message.arbitration_id == PID_REPLY and message.data[2] == ENGINE_RPM:
				rpm = round(((message.data[3]*256) + message.data[4])/4)	# Convert data to RPM

			if message.arbitration_id == PID_REPLY and message.data[2] == VEHICLE_SPEED:
				speed = message.data[3]										# Convert data to km

			if message.arbitration_id == PID_REPLY and message.data[2] == THROTTLE:
				throttle = round((message.data[3]*100)/255)					# Conver data to %

		c += '{0:d},{1:d},{2:d},{3:d}'.format(temperature,rpm,speed,throttle)
		print('\r {} '.format(c))
		print(c,file = outfile) # Save data to file
		count += 1
			

 
	
except KeyboardInterrupt:
	#Catch keyboard interrupt
	GPIO.output(led,False)
	outfile.close()		# Close logger file
	os.system("sudo /sbin/ip link set can0 down")
	print('\n\rKeyboard interrtupt')