linux-can/can-utils #292: candump on Vehicle OBD2 port receiving only 2 Identifiers

Mike20001, 2021-05-06:
I connected a CAN transceiver to my vehicle and when I use candump it gives me a lot of different identifiers each sending different data frames so I am pretty sure there is nothing wrong with my hardware. But when I tried connecting it to another vehicle I am receiving the following output:
can0 45A [08] 5A 04 00 00 00 00 00 00
can0 4E0 [08] 24 00 00 00 00 00 00 00
can0 45A [08] 5A 04 00 00 00 00 00 00
can0 4E0 [08] 24 00 00 00 00 00 00 00
can0 45A [08] 5A 04 00 00 00 00 00 00
can0 4E0 [08] 24 00 00 00 00 00 00 00
can0 45A [08] 5A 04 00 00 00 00 00 00
can0 4E0 [08] 24 00 00 00 00 00 00 00 

As you can see I keep getting the same 2 identifiers and they both send the exact same data each time. But I am sure that this vehicle more ECU's because when I connect my bluetooth elm327 I can see all the information concerning my vehicle (rpm, wheel position, speed...) and it show that it is using the can protocol iso 15765-4.

Is it possible that the can bus that I am connected to on the obd2 connector is a seperate can bus from all the other ECU's in the vehicle and its only purpose is to wait till an obd2 request is sent to it and reply with the desired information?

marckleinebudde, 2021-05-07:
> As you can see I keep getting the same 2 identifiers and they both send the exact same data each time. But I am sure that this vehicle more ECU's because when I connect my bluetooth elm327 I can see [...]

I'm not a car hacker, so I'm just speculating here :)

There is more than a CAN bus on the obd2 connector. There is also K-Line and J1850. Maybe the elm327 has support for these busses, too.

> Is it possible that the can bus that I am connected to on the obd2 connector is a seperate can bus from all the other ECU's in the vehicle and its only purpose is to wait till an obd2 request is sent to it and reply with the desired information?
> 
> Also I know this probably isn't the best place to ask this question but I don't really know of any forum where this question is suitable.

Try searching for you car brand and model and odb2 or elm327. Car hacking seems also be an interesting search keyword.

olerem, 2021-05-07:
Quick search results:
https://en.wikipedia.org/wiki/Unified_Diagnostic_Services
https://github.com/zombieCraig/uds-server
https://github.com/raychow/UDSonCAN/tree/master/UDSonCAN
https://github.com/pylessard/python-udsoncan
https://github.com/pylessard/python-can-isotp