linux-can/can-utils #296: candump process exit automatically when system resume back

Joakim-Zhang, 2021-05-28:
Hi experts,

I found candump process exit automatically when system resume back, we can reproduce it with below sequence:
1) candump can0 &
2) echo mem > /sys/power/state
3) wakeup system

You can see something like below when system resume back:
[ 2955.104619] PM: resume devices took 0.548 seconds
[ 2955.232468] OOM killer enabled.
[ 2955.235618] Restarting tasks ... done.
[ 2955.241259] PM: suspend exit
[1]+  Done                    candump can0

After a quirk check, bisect points to the broken commit, candump can work like before if revert this commit.
commit 639498bc801a22f7fa2e58681eee111eeca87db
Author: Jörg Hohensohn <idc-dragon@gmx.de>
Date:   Thu Dec 3 17:36:32 2020 +0100

    candump: use epoll_wait() instead of select()

    Using epoll_wait() instead of select() gives higher
    performance for listening on multiple interfaces.
    Additionally, the read order has a higher chance
    to resemble the true temporal order.
    select() gives implicit priority to the lower index socket.

    Signed-off-by: Jörg Hohensohn <joerg.hohensohn@gmx.de>

This patch is a big change, and I have no idea how to fix it, so raise a issue here.

Joakim Zhang

marckleinebudde, 2021-05-28:
Hey @Joakim-Zhang,

can you start candump with `strace` and post the output.

IDC-Dragon, 2021-05-28:
Oh, I'm embarrassed.
(Frankly, I wouldn't have expected a candump to work over a suspend, since this is no strong side of Linux in general, or at least embedded Linux...)
I'm thinking about how to reproduce, would need to setup a system capable of suspend to RAM. What kind of system and adapter have you been using? Does it fail when using a virtual interface, too?

marckleinebudde, 2021-05-28:
```
ioctl(4, SIOCGIFINDEX, {ifr_name="can0", }) = 0                                                                                              │
setsockopt(4, SOL_CAN_RAW, 5, [1], 4)   = 0                                                                                                  │
bind(4, {sa_family=AF_CAN, sa_data="\337\266\4\0\0\0\t\0\0\0\320\221\360\266\207\360\226|R&\356\266"}, 24) = 0                               │
epoll_wait(3, 0xbe950328, 1, -1)        = -1 EINTR (Interrupted system call)                                                                 │
close(4)                                = 0                                                                                                  │
close(3)
```

Joakim-Zhang, 2021-05-28:
Hi @marckleinebudde,

~# strace candump can0
execve("/usr/bin/candump", ["candump", "can0"], 0xffffc45b2d48 /* 21 vars */) = 0
brk(NULL)     [  144.344604] can: controller area network core
                          = 0xaaa[  144.349833] NET: Registered protocol family 29
ac54a8000
faccessat(AT_FDCWD, "/etc/ld.so.preload", R_OK) = -1 [  144.360609] can: raw protocol
ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=71196, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 71196, PROT_READ, MAP_PRIVATE, 3, 0) = 0xffffab66a000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\360J\2\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=1432848, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffffab6aa000
mmap(NULL, 1526232, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xffffab4f5000
mprotect(0xffffab64c000, 65536, PROT_NONE) = 0
mmap(0xffffab65c000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x157000) = 0xffffab65c000
mmap(0xffffab662000, 31192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xffffab662000
close(3)                                = 0
mprotect(0xffffab65c000, 12288, PROT_READ) = 0
mprotect(0xaaaabf216000, 4096, PROT_READ) = 0
mprotect(0xffffab6af000, 8192, PROT_READ) = 0
munmap(0xffffab66a000, 71196)           = 0
rt_sigaction(SIGTERM, {sa_handler=0xaaaabf202e20, sa_mask=[TERM], sa_flags=SA_RESTART}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
rt_sigaction(SIGHUP, {sa_handler=0xaaaabf202e20, sa_mask=[HUP], sa_flags=SA_RESTART}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
rt_sigaction(SIGINT, {sa_handler=0xaaaabf202e20, sa_mask=[INT], sa_flags=SA_RESTART}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
epoll_create1(0)                        = 3
socket(AF_CAN, SOCK_RAW, CAN_RAW)       = 4
epoll_ctl(3, EPOLL_CTL_ADD, 4, {events=EPOLLIN, data={u32=3206639824, u64=187650327802064}}) = 0
ioctl(4, SIOCGIFINDEX, {ifr_name="can0", ifr_ifindex=4}) = 0
setsockopt(4, SOL_CAN_RAW, 5, [1], 4)   = 0
bind(4, {sa_family=AF_CAN, sa_data="\377\377\4\0\0\0p\0k\253\377\377\0\0\0 k\253\377\377\0\0"}, 24) = 0
epoll_pwait(3,

marckleinebudde, 2021-05-28:
The interesting part happens to `epoll_wait()` _after_ the resume.

But I've already created a trace, see https://github.com/linux-can/can-utils/issues/296#issuecomment-850293955

I'm currently preparing a fix.

marckleinebudde, 2021-05-28:
@Joakim-Zhang

please test https://github.com/linux-can/can-utils/pull/297

closed 2021-05-28 by marckleinebudde