linux-can/can-utils #298: candump can0 does not show data

Niki2896, 2021-06-02:
can someone help out with this similiar problem?

We have a Nvidia Jetson Nano (UBUNTU 18.04.LTS) connected to a USB-CAN Adapter (Peak). The Adapter is linked via CAN to a CVMU (Cell Voltage Monitoring Unit) to observe the voltages of a small fuell cell stack.
We set up the adapter by typing the following into the terminal:

    $ sudo apt-get install can-utils
    $ sudo ip link set can0 up type can bitrate 500000
    $ sudo ip link set can0 up (Adapter control LED signals that it is now connected (1Hz blinking))
    $ ifconifg (can0 is displayed)

When we now command in the terminal
$ candump can0
the curser jumps on the next line (no more directory) and nothing happens anymore. Also the Adapter LED does not signal data transfer (should be 2Hz blinking)
We have been not able to find out if this is related to a software or hardware issue. If someone can give us useful hints, we would appreciate it! Question is, if the terminal shows nothing because there is no data transfer and the cables have to be checked or if there are further interface configurations in Linux necessary.
Kind regards

marckleinebudde, 2021-06-02:
* Are you using the correct bitrate?
* Is your CAN bus properly terminated?
* Is your CVMU supposed to sending CAN messages?
* Which peak adapter are you using?

blakewford, 2021-09-14:
I am noticing similar command line behavior after upgrading from Ubuntu 18.04. I think that puts the changes between kernel versions 5.4 and 4.15. My setup was working prior to the upgrade and still works now with a virtual CAN device. I am using a Microchip USB CAN Analyzer but when I switch to using that device the behavior appears. Not sure what to do with that info but I thought I would add to the conversation here.

marckleinebudde, 2021-09-15:
Which driver are you using? `drivers/net/can/usb/mcba_usb.c`?

blakewford, 2021-09-15:
Yes, exactly.

marckleinebudde, 2021-09-15:
Which _exact_ kernel versions are working and not working?

blakewford, 2021-09-15:
It is hard to be exact with the working version ~4.15 after my upgrade. Non-working at 5.11.0.

marckleinebudde, 2021-09-15:
Try latest v5.11.x. In https://github.com/linux-can/can-utils/issues/298#issuecomment-919578345 you've written v5.4 is not working...

marckleinebudde, 2021-09-15:
Can you try a `git bisect`?

```
git bisect start v5.4 v4.15
# compile kernel and test
git bisect bad # if it's not working
git bisect good # if it's working
# compile, rinse, repeat
```

blakewford, 2021-09-18:
I updated the above comment. I wanted to preserve the context, but express more how wild my initial guess was. The latest kernel version numbers are the good ones so I am reposting the requested work with that in mind. In my current situation I have a functional workaround so I was just posting my problem here to up vote a fix. I really don't have the time to do this work but if someone else does I would be grateful. 

git bisect start v5.11 v4.15
# compile kernel and test
git bisect bad # if it's not working
git bisect good # if it's working
# compile, rinse, repeat

marckleinebudde, 2023-10-03:
What's your workaround?

closed 2023-10-03 by marckleinebudde