linux-can/can-utils #303: Problem with receiving CAN bus data using candump can0

dzeko1234, 2021-06-29:
Hello ,
I am using RPi 3 with Raspbian distribution ,socketCAN driver and PICAN2 board connected to a Motor control board .

I set the can0 up by this command :

sudo /sbin/ip link set can0 up type can bitrate 1000000 .

When i send a CAN frame using cansend can0 it works fine , i receive the frame sent in my control board .

The control board send a response frame to pican2 .

My problem is that candump can0 did not receive the response frame .

Currently I have 2 terminal , 1 for sending and 1 for receiving .

what’s wrong !!! I can’t find the problem .

Thank you

marckleinebudde, 2021-06-29:
Hey @dzeko1234

What's your `candump` command line? (please paste test, no screenshots).
What does `candump` show if you're sending the CAN command with `cansend`?

Marc

dzeko1234, 2021-06-29:
My command is  
candump can0 

I must receive a response frame from the control board .
candump didn't show anythings .
BTW when I tested on the loopback mode it shows the frame sent .

marckleinebudde, 2021-06-29:
Please start `candump any,0:0,#FFFFFFFF -extA` and on a different terminal do your `cansend`. You should see `TX` message. If not then check your CAN bus cabling, proper termination and same bitrate on both sided.

dzeko1234, 2021-06-29:
@marckleinebudde  thank for your reply .
I tried your solution , the problem persist . No received data when i run candump any,0:0,#FFFFFFFF -extA .
I configured the same bitrate  and cabling is well.

I didn't understand why pican2 send frame fine but didn't receive .

I tested on another node . I send 101#1111 from the node to my pican2 . Pican2 receive the data but display it many time , second in the node i get  " an error passive "    

I am bolcked since 4 day , i can't solve this . please help me

marckleinebudde, 2021-06-29:
Some background information:
A CAN frame send on the bus must be ACKed by at least one other CAN controller on the bus. If the sender doesn't see the ACK it repeats the CAN frame and the TX error counter is increased. This is repeated until the CAN frame is ACKed and the error counter will reach error passive status.

This means, if your second node goes into error passive and the first node receives the CAN frame many times, there is something wrong with the first node. Make sure the CAN interface is not configured as "listen only" in Linux, and make sure the transceiver is properly turned on.

dzeko1234, 2021-06-29:
@marckleinebudde  I have another problem . It seem that the response from sent from the control board is FD . 
So i must set  the mode Fd On .
When i run this
sudo /sbin/ip link set can0 up type can bitrate 1000000 dbitrate 1000000 fd on 

or this 
sudo /sbin/ip link set can0 up type can bitrate 1000000 dbitrate 1000000 fd-non-iso on 

I get "  Operation not supported"

Can you suggest how to solve this . thank you

marckleinebudde, 2021-06-29:
The pican2 doesn't support CAN-FD, at least if it's this board: https://copperhilltech.com/pican-2-can-bus-interface-for-raspberry-pi/

dzeko1234, 2021-06-29:
Yes It's

marckleinebudde, 2021-06-29:
The mcp2515 doesn't support CAN-FD. You need a mcp2518fd based board, e.g. https://www.seeedstudio.com/CAN-BUS-FD-HAT-for-Raspberry-Pi-p-4742.html. Note you should avoid the mcp2517fd, which is broken.

marckleinebudde, 2021-06-29:
Can we close this issue?

closed 2021-06-29 by dzeko1234