linux-can/can-utils #304: No Data on OBD using candump/sniff

mB-PiBox, 2021-07-05:
Im using a Raspberry Pi 4 and the [Waveshare 2Ch CAN Hat](https://www.waveshare.com/2-ch-can-hat.htm) trying to tap into my trucks OBD port. 

I can wire the can1 (low) to can0 (low) on the board, then issue a cansend on can1 while running candump on can0. Doing this I’m able to see the data being sent in the terminal window running the dump. 

However once I connect can0 to my trucks OBD (pin 14 for CAN_L) I cannot see any data when doing a candump or cansniff. I’ve tried different bitrates (125000, 250000, 500000) but still get nothing. I also tried CAN_H (pin 6) but no data there either.. 

What am I missing?

marckleinebudde, 2021-07-05:
Hi @mB-PiBox,

CAN is a two wire bus, you have to connect both CAN high and CAN low. If it still doesn't work, I suggest to look for a car or truck hacker forum and read/ask for details there. Maybe you have to send a CAN message to enable CAN output on the obd port.

regards,
Marc

mB-PiBox, 2021-07-06:
> Hi @mB-PiBox,
> 
> CAN is a two wire bus, you have to connect both CAN high and CAN low. If it still doesn't work, I suggest to look for a car or truck hacker forum and read/ask for details there. Maybe you have to send a CAN message to enable CAN output on the obd port.
> 
> regards,
> Marc

Sorry for not properly detailing the situation. Both CAN_L & CAN_H are connected the can0 (as well as Signal Ground from the OBD port).. I was just using CAN_L as example in my post. 

I get data on the can when using a CAN scan tool (separate from the Pi/Hat) without having to send data down the line.. its just the PiHat w/ can-util that doesn't seem to be working.

hartkopp, 2021-07-06:
For OBD you need to set the bitrate to 500000 bit/s - and then set the CAN interface to 'up'.

Can you please send the output of `ip -det link show can0` then?

olerem, 2021-07-07:
OBD port is not directly connected to any network segment. It is connected to a gateway. To get some response, you will need to send a request, for example a request from OBD scanner

hartkopp, 2021-07-07:
> OBD port is not directly connected to any network segment. It is connected to a gateway. To get some response, you will need to send a request, for example a request from OBD scanner

Yes! It would make sense to let the OBD scanner do his normal work and attach the CAN bus from the CAN-HAT in parallel to watch the CAN traffic. This makes sure that at least the termination and the bitrate is correct.