linux-can/can-utils #305: no viable conversion from 'QCanBusFrame' to 'qreal'

dzeko1234, 2021-07-06:
Hello 
I want to display frame received from Can bus in my Qt/c++ App .
 
I use this code :
QCanBusFrame framerpm;
    framerpm.setFrameId(101);
    QByteArray payloadrpm("0x10001800");
    framerpm.setPayload(payloadrpm);
    device->writeFrame(framerpm);
    QCanBusFrame rpmframe = device->readFrame();
    back.setValue(framerpm); // I want to convert this framerpm to Setvalue which is qreal . the SetValue will be displayed in a Qml radialBar using Signal and slots .
 Does any one know how to solve this .
thankyou

hartkopp, 2021-07-06:
Sorry, but this has nothing to do with can-utils.
Please ask for help in a Qt forum.

closed 2021-07-06 by hartkopp

dzeko1234, 2021-07-14:
Hello , 
Currently i am developping a Qt application wich send and receive Can frame .
 
I use the socketCan driver and can0 interface .

I have problem in getting  the current connected device  " std::unique_ptr<QCanBusDevice> m_canDevice " from class to other .

My question is :  if i create another can device  using  "QCanBusDevice *device = QCanBus::instance()->createDevice(
                QStringLiteral("socketcan"), QStringLiteral("can0"));"    it cause probelms ?

marckleinebudde, 2021-07-14:
I don't know what the `QCanBus` does behind the scenes, but you can open more than one CAN socket, even on the same interface, in a single application.