linux-can/can-utils #306: Cansniffer freezing for about 5 seconds when filtering ID's using bitmask after runtime

mitchdetailed, 2021-07-13:
i've recently noticed that cansniffer is freezing for a few seconds after adding or removing frames using a bitmask. (+500700) or (-1237FF), etc.

marckleinebudde, 2021-07-14:
Hey @mitchdetailed,

can you give me details:
* Which kernel version are you using?
* Which can-utils version are you using?
* What are the steps to reproduce the issue on the command line?

regards,
Marc

hartkopp, 2022-05-04:
Hm @mitchdetailed - this issue is open for a while now.

Can you tell us the can-utils version, Marc was asking for? Did you get the can-utils via Debian/Ubuntu package (which is unfortunately pretty outdated) or did you build it on your own from this repository?

mitchdetailed, 2022-05-18:
just the common package using sudo apt-get. i can confirm that it still is
an issue when using masks. youtube video link will be supplied shortly

mitchdetailed, 2022-05-18:
https://youtu.be/rJkoJQktLWs

hartkopp, 2022-05-18:
Thanks for the video! Never seen such behavior!

So still the questions from Marc stay open:

What is you Linux kernel version? (output of `uname -a`)
What is your can-utils version? (output of `dpkg -l can-utils`)

hartkopp, 2022-05-19:
Hi @mitchdetailed ,

I think I found the problem.

Since this commit https://github.com/torvalds/linux/commit/d5f9023fa61ee8b94f37a93f08e94b136cf1e463 the removal of CAN_BCM jobs is delayed to remove some data structures in a safe way. I've never seen this effect but with the use of the 'old' BCM-based `cansniffer` which is now only available here https://github.com/linux-can/can-tests/blob/master/bcm/cansniffer.c the delay of removing 2048 BCM jobs can be shown easily.

So this safety fix of the BCM obviously needs to be revisited in term of performance :-/

To fix the problem for **you**:

1. remove the out-dated can-utils package from your system with `apt remove --purge can-utils`
2. clone the latest can-utils from this repository `git clone https://github.com/linux-can/can-utils.git`
3. `cd can-utils`
4. `make`
5. `sudo make install`

The new `cansniffer` does not use the CAN_BCM but the CAN_RAW sockets (and supports 29bit CAN IDs).

mitchdetailed, 2022-05-19:
in case nobody has told you today, you're awesome Oliver!

hartkopp, 2022-05-19:
> in case nobody has told you today, you're awesome Oliver!

Thanks Mitch!

But as you can see, improvement can only happen when people like you use that stuff and provide valuable bug reports (including your eye-opening video!). So I can give that **_'thank you'_** back to Tennessee !!