linux-can/can-utils #309: get stuck for 10ms every 10~11 frames send

swordjjjkkk, 2021-07-26:
i use can-j1939 send 256bytes data, it spend a long time .
i check can-dump found that
[image]
every 10~11 frame, it stuck for about 10ms.
here is my send code.
[image]
it that normal or a problem？

olerem, 2021-07-26:
Hi @swordjjjkkk ,

> i use can-j1939 send 256bytes data, it spend a long time

Do you expect some different speed? Do you have some comparison numbers?

> every 10~11 frame, it stuck for about 10ms. it that normal or a problem？

It is normal, it is withing spec and limited to your system configuration.

On the send code path, there is two buffers:
1. socked buffer, which is usually 300K. If you would use MSG_DONTWAIT, then your sendto would return immediately.
2. the interface buffer. This one is usually limited to ~10 packets by default. The j1939 stack is trying to push as many packets as possible, If interface buffer is full, the stack is starting 10ms+some random milliseconds timer and trying to push more data after timeout is reached.

You can increase the interface buffer, but this would potentially affect other local and bus senders. So, other applications would have higher latency, but the j1939 transfer would have no gaps.

marckleinebudde, 2021-07-26:
Hello @swordjjjkkk,

What's your CAN bus speed?
Which CAN controller are you using?
Can you create a CAN dump on the receiving side?

swordjjjkkk, 2021-07-26:
> Hi @swordjjjkkk ,
>
>
> Do you expect some different speed? Do you have some comparison numbers?
>
>
> It is normal, it is withing spec and limited to your system configuration.
>
> On the send code path, [...]

thank you for your reply, that's exactly what i need to know.
i want to test time delay（i think it better transfer with no gaps），so how can i increase the interface buffer?

marckleinebudde, 2021-07-26:
`ip link set can0 txqueuelen 10`

With `10` being the default. Try bigger values, something like `1000`.

swordjjjkkk, 2021-07-27:
it works, thank you all!!

closed 2021-07-27 by swordjjjkkk