linux-can/can-utils #312: testj1939: socket(j1939): Invalid argument.

miteshdube, 2021-08-24:
I am following this [documentation](https://github.com/linux-can/can-utils/blob/master/can-j1939-kickstart.md). But for `modprobe` I'm getting below error.
```
~ # modprobe can-j1939
modprobe: FATAL: Module can-j1939 not found in directory /lib/modules/4.4.207
```
`testj1939` command shows below error. 
```
~ # testj1939 -B -r can0
testj1939: socket(j1939): Invalid argument
```

I think this error is due to not loading of kernel module for `can-j1939`, if I am correct then can anyone please provide me source code for `can-j1939`.

marckleinebudde, 2021-08-24:
j1939 is included in the Linux kernel since v5.4. Consider updating to a newer kernel.

miteshdube, 2021-08-30:
@marckleinebudde Can we back port the CAN-j1939 driver from Linux5.4 to Linux 4.4.207?

marckleinebudde, 2021-08-30:
Hey @miteshdube,

sure, it's Open Source. However, using the latest mainline kernel is preferred.

prajwal-mulkalwar, 2022-01-29:
Facing same issue in Jetson Nano
4.9.201-tegra

closed 2022-01-31 by marckleinebudde