linux-can/can-utils #322: shutdown()

SeifHadrich, 2021-11-16:
I am looking for for a solution to shutdown the can socket properly, but when using shutdown() I got an error : "Operation not supported"
So it looks like the shutdown routine is not supported in the can socket.

marckleinebudde, 2021-11-16:
Calling `close()` should be sufficient.

SeifHadrich, 2021-11-16:
I want to fore the read() to exit and stop waiting for coming msg
close() wasn't a consistent solution, some times the read() will quit and sometime it didn't.
Sending a poison message it's an other solution but it adds complexity to the system

marckleinebudde, 2021-11-17:
I see. You have another thread blocking on the `read()` call? This functionality is not implemented. That needs changes to the kernel, I think that's not too complicated. Is patching and recompiling the kernel an option for you?

hartkopp, 2021-11-17:
Isn't `select()` the right way to handle this?

marckleinebudde, 2021-11-17:
Depends on your programming model :)

SeifHadrich, 2021-11-17:
> I see. You have another thread blocking on the `read()` call? This functionality is not implemented. That needs changes to the kernel, I think that's not too complicated. Is patching and recompiling the kernel an option for you?

Yes it is, and yes patching and recompiling the module sounds it is a option for me.

SeifHadrich, 2021-11-17:
> Isn't `select()` the right way to handle this?

The way my program is designed to use the read() in block mode, so we don't need to re-assemble the frame.
Thus I think select() isn't a good option for this case.

SeifHadrich, 2021-11-17:
> You will always get a single frame with `read()`. You can use `recvmmsg()` to receive multiple frames with one system call.

Yes but when I use read() with no blocking and set a timeout, the read() will be forced to return even if the whole frame is not fully received right ?

marckleinebudde, 2021-11-17:
You will get a single frame with `read()` or no frame at all, independent of blocking or nor, or timeouts.

SeifHadrich, 2021-11-17:
> You will get a single frame with `read()` or no frame at all, independent of blocking or not, or timeouts.

@hartkopp and @marckleinebudde
=> That's sounds as a great solution for this issue.
Thanks a lot, I appreciate your support :)

**Summary and conclusion for this issue  :**
 1- close() the socket is not enough to wake up the read() which is running in thread() and configured to block till recieve a new -message
 2- shutdown() should be the right choice to wake up the read(), but it's not implemented in the current version of the can socket.
 => The solution will be using read() with no block and check if was returning because it received a new frame or because of timeout.  (read() will never return an incomplete frame so returns only a full frame or nothing).

closed 2021-11-17 by marckleinebudde