linux-can/can-utils #323: candump / canmonitor: Survive interfaces going up and down

mickeyl, 2021-11-28:
I'd love to see `candump` and `canmonitor` to a) also launch when starting with an interface that is DOWN and b) survive the interface going down and up again.

Backstory: I'm working on a custom CAN adapter based on Linux which exposes its functionality through various interfaces and when clients open a channel it needs to bring the interface temporarily down to (re)configure the bitrate. It's pretty unconvenient to relaunch `candump` just because someone reconfigured the interface.

hartkopp, 2021-11-28:
Did you already try the '-D' option of `candump` ?

mickeyl, 2021-11-28:
My bad: I read about it in the manpage, then tried using `candump -cataD`, appending it to my favorite bunch of options, only to think it wouldn't be implemented yet when it didn't work -- not realizing that appending it to the `ta` option would not work.

Anyways, with `candump -D -cata`, everything is fine now. Thanks for this greatt set of tools! (and sorry for the noise)

closed 2021-11-28 by mickeyl

hartkopp, 2021-11-28:
In fact I had to look for that option myself :-D

And finally found a typo ... https://github.com/linux-can/can-utils/commit/a705182b2894c700ad83c04a720dc8a345861d01 

So your heads up had a valuable result!

Have fun!