linux-can/can-utils #332: How to connect to usbcan from  the  docker container (which is running on MacOS)?

dodoru, 2022-01-05:
After read this issue(https://github.com/linux-can/can-utils/issues/27)
I got the tips from @marckleinebudde : `The kernel on the macOS doesn't support socketcan. You have to use a linux machine or port socket-can to the macOS kernel or write a slcan userspace driver.`

Now it is 2022.
I have build docker containers base on Ubuntu and install the can-utils.
But I don't know how to connect to usbcan from the docker container(insided) ?

marckleinebudde, 2022-01-05:
I rephrase your use case, to be sure that I understand it correctly :)

- Your host system is OSX.
- You want to run an Ubuntu based Docker container.
- You have an USB based CAN adapter, which is supported by Linux.
- You want to use the USB based CAN adapter inside the Docker container.

If that is your use case, the task is to add the USB device to a Docker container on OSX. This is independent of socketcan.

The first hit looks reasonable: https://www.google.com/search?q=add+usb+port+to+docker+container+on+osx&oq=add+usb+port+to+docker+container+on+osx

rusoku, 2022-01-05:
Another solution is to use mac-can:
https://github.com/mac-can

marckleinebudde, 2022-01-05:
IMHO https://github.com/mac-can solves a different use case...

AltayAkkus, 2022-02-12:
IMO @marckleinebudde 's link is the most promising solution.
But: The articles talk about integrating the hardware devices into the Docker host/machine.
[image]
Then you could use it in your containers.
So technically, you are connecting the device to your Container, but rather to the parent host. 
I dont know if that meets your goal, it would mean that a simple ```docker pull xyz/xyz``` would not be enough to deploy your application.

The other problem is that while I searched for "add USB device to docker" I stumbled upon dozens of threads that ended with no success, and also the command line utility mentioned in these [articles](https://christopherjmcclellan.wordpress.com/2019/04/21/using-usb-with-docker-for-mac/) are deprecated, and Docker seemed to get a major rework on Mac OS X too.
If I were you, I would try to run it on a bare metal linux machine, even if its just a Raspberry Pi, or something.

(And I agree, the problem is more Docker heavy, maybe try asking there)