linux-can/can-utils #344: No error recovery on CANUSB

zevv, 2022-02-14:
Hi, I'm having some troubles with a Lawicel CANUSB interface that does not seem to recover from a particular error case. 

TL;DR: CAN interface goes defunct when sending data on an bus with electrical problems, and does not recover.

Long version:

Setup:
- Setup a simple CAN bus with one device and a linux machine with CANUSB interface
- attach CAN interface using `slcan_attach -s5 <TTY>`
- start slcand with `slcand -o -c -f -s5 <TTY>`
- bring up link: `ip link set qlen 255 slcan0 up`
So far so good, `candump` shows frames from the external device and I'm able to send on the bus using `cansend`

Error situation with that recovers:
- Short circuit CAN bus, physically
- No more frames are seen on the bus with `candump`. My particular device goes into bus-off and recovers
- Remove short circuit
- Frames are again received on the bus

Error situation from which I can not recover:
- Short circuit CAN bus, phsyically
- Send one or more frames on the bus using `cansend`
- Remove short circuit
- Bus remains defunct: no more frames can be seen on the bus, `candump` shows no data

Unfortunately the slcan driver does not seem to support `restart` and `restart-ms` `ip` commands. The only way to recover the bus is by detaching and reattaching the bus and restarting `slcand`.

Manually connecting to the CANUSB interface after the problem occurs, using the `F` command can be used to get the status flags. Return code in this case is 0xA4, indicating "error passive" and "bus error", according to the CANUSB manual.

My questions:
- Can this problem be avoided somehow?
- Is there a proper way to programatically recover from this defunct bus?
- Is the CANUSB considered a production grade interface, or should it only be used for hobby & development?

Thank you,

zevv, 2022-02-14:
FWIW: a little experimentation on the low level serial protocol shows that the error state is indeed indicated by the flag command `F` returning `0xA4`. Reading the flags again after that returns `0x00`, but the error situation does not resolve. Only when first closing and then reopening the CAN channel using `C` and `O` commands the interface becomes responsive again.

marckleinebudde, 2022-02-14:
> Can this problem be avoided somehow?

Do I understand you correct, the difference between the device automatically recovers and does not recover situation is that you send a CAN frame after the CAN bus has been shorted?

The device probably goes into bus off because it's trying to send and doesn't recover from this.

> Is there a proper way to programatically recover from this defunct bus?

Is there a way to detect if the device is in bus off? Does the device send you a I'm-in-bus-off message, or do you have to poll with the `F` command for this?

If you can detect the bus off, then the driver can be enhanced to support `restart` and `restart-ms`.

> Is the CANUSB considered a production grade interface, or should it only be used for hobby & development?

I would never use any device with a Lawicel compatible firmware, not even for hobby projects, for above reasons. If you want a USB to CAN adapter, there are plenty of `gs_usb` compatibles out there. For example the Open Hardware/Open Source CandleLight, e.g. https://shop.linux-automation.com/candlelight-D02-R01-V01-C00-en

zevv, 2022-02-14:
> Do I understand you correct, the difference between the device automatically recovers and does not recover situation is that you send a CAN frame after the CAN bus has been shorted?

Right.

> The device probably goes into bus off because it's trying to send and doesn't recover from this.

This seems to be the case, indeed. In theory, this could be solved (worked around) at the driver level: by periodically sending out the `F` command and getting the results the error state could be detected, after which a `C` close and `O` open should be sent to get the device to perform a bus recovery.

Anyway, it seems that we should look into an alternative interface; not great news as we just ordered 40 lawicels, but I guess we should be able to return those.

Thanks!

marckleinebudde, 2022-02-14:
Doh! Better cancel the order. Drop me a note, maybe I can get you some discount for the CandleLights...

marckleinebudde, 2022-02-14:
> This seems to be the case, indeed. In theory, this could be solved (worked around) at the driver level: by periodically sending out the F command and getting the results the error state could be detected, after which a C close and O open should be sent to get the device to perform a bus recovery.

ACK. If the device doesn't actively send it's status you have to poll...Is not that complicated to setup a worker under Linux to do this.

tomatghub, 2022-02-15:
CAN Standards detail how CAN controllers should handle and recover from BUS-OFF. Which is to wat for 127 (or so) idle bits before recovering. This can optionally be controlled by software (if you don't want it to recover automatically). But they have to recover somehow.

The  Lawicel CANUSB uses an SJA1000 controller chip. The Data Sheet for that chip ... doesn't say how it handles BUS-OFF Recovery at all. Fortunately there's an NXP App Note that does. Section 5.2 of this:

https://www.nxp.com/docs/en/application-note/AN97076.pdf

> As shown in Figure 25 bus-off can only be terminated with the host controller command  ́Reset Request =
> 0 ́. This will start the bus-off recovery time

Until that command is sent, the SJA1000 will remain locked and can't communicate. It needs "permission" to follow the standard BUS-OFF recovery operation.

That means the controller in the device should perform that action, or should make that condition and command visible to the driver. You shouldn't have to close/open the device to perform that operation. The Lawicel CANUSB Manual says "Bus Error (BEI), see SJA1000 datasheet", but doesn't say how to recover from this.

marckleinebudde, 2022-02-15:
Thanks for the insight @tomatghub, so it all boils down that the Linux driver needs to know the state of the device. If there is no notification form the device, the driver has to poll for it.

tomatghub, 2022-02-15:
If the Reference Manual for this device doesn't detail a command it can be sent that then sends the correct command to the SJA1000, then it looks like this device CAN'T recover from "Bus Error". Except (apparently) by turning a Red LED on the device on to tell the user to MANUALLY unplug it.

So unless the driver can be written to have control over a small robot to unplug and replug the USB port, then I don't think the driver CAN be written to recover from this.

If this device doesn't support recovery from "Bus Off" then maybe Linux shouldn't even try to support this device until it is changed so it can recover.

zevv, 2022-02-16:
> Please prove me wrong if there is a command like that.

Some manual experiments with the serial protocol show that a `C` close followed by an `O` open seems to do the job.

tomatghub, 2022-02-16:
That's as bad as unplugging. Performing "Close" then "Open" is pretty much a "turn it off and on again" fix. That doesn't follow the requirements for CAN Bus Off Recovery, as well as risks missing packets it should be receiving.

Can someone ask Lawicel if there's a better way to do this? If there isn't, maybe report it is a 25 year old bug and ask them to fix it?

If the device can't follow standard THIRTY ONE YEAR OLD CAN Standards, then it should be fixed so it does, or support for it should be dropped until it does.

This device is based on their 1997 "CAN232" device that was "Improved" in 2003.

zevv, 2022-02-16:
> That's as bad as unplugging [...] missing packets it should be receiving.

I wouldn't take it this far; there is no detaching the serial device from the USB bus, it is just sending some commands over the pipe. I do not see how this would be practically different from any hypothetical command that Lawicel might have put in the do a explicit bus recovery. Also, packets were already lost when the device went into bus-off mode - bus off doesn't "just happen", there's a good reason for the CAN interface to decide it's better to back down and shut up for a while: things are broken on the bus.

if I find time and motivation I can probably put this in the driver, but for the current project I feel the best way for us is to move away from this interface and use something more robust.

Thanks

tomatghub, 2022-02-17:
I've just learned something about CAN. ISO11898 in 6.15 says "Bus Off" devices aren't allowed to receive any frames. Then it contradicts that in 13.1.4.4 by saying they can and it is "Implementation Dependent". It also says they must not auto-recover from Bus Off (like the SJA1000 works) and then it says they can (like many other controllers do). That's what happens when specifications are written after working chips have been in use for years. So for some chips, a "Reset" to recover from Bus Off is an acceptable (even necessary) thing to do.

The best reference for an SJA1000 driver is "sja1000.c" in the linux sources. It gets the  error Interrupt, sees "SR_BS" and performs the normal CAN error recovery (using the configured recovery time). Then that soft-resets the SJA1000.

For this device that means polling for the status. A dedicated "reset bus off" command in the protocol would be nice, but according to the "can/Kconfig" file there are multiple manufacturers of devices using the "LAWICEL ASCII protocol" so that's not going to happen.

As long as the "c" command doesn't reset the chip (losing the baud rate, modes and acceptance filters), then that should work. If some devices (from some manufacturers) do reset the chip, then this mightn't be a workable solution. The manual doesn't say if "c" has side effects, but it doesn't say it doesn't either.

Yes, use a different controller if you can.

marckleinebudde, 2022-02-17:
> Yes, use a different controller if you can.

ACK

> As long as the "c" command doesn't reset the chip (losing the baud rate, modes and acceptance filters), then that should work. If some devices (from some manufacturers) do reset the chip, then this [...]

For historical reasons a lot of configuration of the driver is done by the use space tool. I think the driver doesn't know the configured bit rate and other parameters, so if the chip loses the bit rate configuration during a close/open cycle, this even needs more changes :(

I'm going to close this issue, objections?

zevv, 2022-02-17:
Fine with me, the root problem is clear and we will likely be moving to another controller anyway.

closed 2022-02-17 by marckleinebudde