linux-can/can-utils #348: asc2log wrong timestamps

xR3b0rn, 2022-02-23:
When using `asc2log` for an .asc-logfile generate by BusMaster, the generated .log-file will have really weird timestmaps.

This is the head of the asc:
```
date Mon Dec 6 11:21:13 am 2021
base hex  no internal events logged
// version 7.1.0
40874.808200 1 122 Rx d 8 E3 0C 00 FE 07 00 00 28
40874.809100 1 3E5 Rx d 8 E6 2A 00 00 00 00 00 00
40874.810300 1 040 Rx d 8 D7 0B 00 04 C1 00 00 00
40874.810800 1 16A954A6x Rx d 8 00 00 20 79 78 5A 5A A5
40874.811100 1 0AD Rx d 8 F7 EA BF FF 00 14 00 00
40874.811300 1 0AE Rx d 8 88 0A 00 00 FF 00 FD 81
40874.811600 1 0B1 Rx d 8 44 24 00 00 00 00 00 00
```

This is the head of the .log-logfile:
```
(81748.811600) can0 000#2C18000000000000
(122623.623600) can0 003#00003618003C0132
(163498.437200) can0 000#2A05000000000000
(204373.252500) can0 000#0000000000000000
(245248.077600) can0 003#0E0A080000285028
(286122.903100) can0 000#0000000000000000
(326997.728900) can0 056#1F08160000000200
(367872.555700) can0 003#0300000000001E00
(408747.383000) can0 01E#0A09140008000000
```

This makes replaying the log file impossible (if you want to consider timestamps). Is this a bug in the `asc2log` or in the BusMaster?

xR3b0rn, 2022-02-23:
The commands I've executed:
```
asc2log -I my.asc -O my.log && head -10 my.asc && echo "====================================" && head -10 my.log
```
The output:
```
Setting locale to 'en_US' failed!
Not able to determine original log file date. Using current time.
date Mon Dec 6 11:21:13 am 2021
base hex  no internal events logged
// version 7.1.0
40874.808200 1 122 Rx d 8 E3 0C 00 FE 07 00 00 28
40874.809100 1 3E5 Rx d 8 E6 2A 00 00 00 00 00 00
40874.810300 1 040 Rx d 8 D7 0B 00 04 C1 00 00 00
40874.810800 1 16A954A6x Rx d 8 00 00 20 79 78 5A 5A A5
40874.811100 1 0AD Rx d 8 F7 EA BF FF 00 14 00 00
40874.811300 1 0AE Rx d 8 88 0A 00 00 FF 00 FD 81
40874.811600 1 0B1 Rx d 8 44 24 00 00 00 00 00 00
====================================
(81748.811600) can0 000#2C18000000000000
(122623.623600) can0 003#00003618003C0132
(163498.437200) can0 000#2A05000000000000
(204373.252500) can0 000#0000000000000000
(245248.077600) can0 003#0E0A080000285028
(286122.903100) can0 000#0000000000000000
(326997.728900) can0 056#1F08160000000200
(367872.555700) can0 003#0300000000001E00
(408747.383000) can0 01E#0A09140008000000
(449622.214000) can0 002#00003C1B00000000
```

hartkopp, 2022-02-23:
> The command I've executed:
> 
```
# asc2log -I my.asc -O my.log && head -10 my.asc && echo "====================================" && head -10 my.log
Setting locale to 'en_US' failed!
Not able to determine original log file date. Using current time.
```

The problem is here!

> date Mon Dec 6 11:21:13 am 2021
> base hex  no internal events logged

This line does not match the Vector specification!
There is no `timestamps <absolute|relative>` identifier.

Example:
```
date Sat May 16 12:54:09 2020
base hex  timestamps absolute
no internal events logged
```

Adding `timestamps absolute` in your logfile as shown above fixes your issue.

closed 2022-04-08 by hartkopp