linux-can/can-utils #349: When ISO-TP Socket is not bound call to write sets errno EADDRNOTAVAIL and not EBADF

derek-will, 2022-03-09:
When CAN_ISOTP is not bound, then a call to `write` sets errno  **EADDRNOTAVAIL** and not **EBADF** as I would expect per the [documentation](https://man7.org/linux/man-pages/man2/write.2.html).

I would expect **EADDRNOTAVAIL** to be returned when passing in nonsense address information into `bind` which according to the [source code](https://github.com/torvalds/linux/blob/ffb217a13a2eaf6d5bd974fc83036a53ca69f1e2/net/can/isotp.c#L1099) it does.

Is there a rationale for this errno on `write` when the socket is not bound?

derek-will, 2022-03-10:
After doing some thinking and experimenting, for UDP sockets one would get **EDESTADDRREQ** if a `write` is attempted without a `connect`. However, CAN_ISOTP has no support for `connect` and so therefore **EADDRNOTAVAIL** is the next best thing. The whole "Cannot assign requested address" comment just threw me as unexpected and it is typically an errno associated with `bind`. Something like **EBADF** would adhere to the manpage for `write` and "File descriptor in bad state" would sort of make sense.

On the other end, when trying to do a `read` on a UDP socket without calliing `bind` first then the call will block indefinitely. CAN_ISOTP exhibits the same behavior. It is sort of undesirable. Now this is likely outside the scope of SocketCAN, but it would be nice if the sockets API would return an error and set an appropriate errno in such a state, but it appears that this is not the case. I am always a big fan of "fail early" and it seems like the sockets API does not do this in this case.

hartkopp, 2022-03-10:
> On the other end, when trying to do a `read` on a UDP socket without calling `bind` first then the call will block indefinitely. CAN_ISOTP exhibits the same behavior. It is sort of undesirable. Now this is likely outside the scope of SocketCAN, but it would be nice if the sockets API would return an error and set an appropriate errno in such a state, but it appears that this is not the case. I am always a big fan of "fail early" and it seems like the sockets API does not do this in this case.

Oh, that's an interesting idea.
If I understand you correctly you suggest to return  EADDRNOTAVAIL when reading from an unbound isotp socket?

derek-will, 2022-03-14:
> Oh, that's an interesting idea.
> If I understand you correctly you suggest to return EADDRNOTAVAIL when reading from an unbound isotp socket?

Thanks for filtering through those thoughts. :-) 

Yes, I think it would be best if when someone attempts to read from an unbound ISO-TP socket that they get rejected and the appropriate errno is set. I do not think there is a case or there ever would be one where an application developer would need to read from an unbound ISO-TP socket.

For example, with the J2534 API a call to `PassThruReadMsgs` will return **ERR_NO_FLOW_CONTROL** on a ISO15765 channel without a [filter](http://www.drewtech.com/support/passthru/startmsgfilter.html) set.

If reads on unbound sockets will never succeed, then rejecting the call to `read`, `recv`, etc. and setting an appropriate errno will assist the application developer in tracing errors in their code. I think EADDRNOTAVAIL is a sensible choice.

derek-will, 2022-03-15:
Hi Oliver, I emailed you my email address. What a strange statement. :-)

hartkopp, 2022-03-17:
Hi Derek,

your suggestion is now in the upstream process and will show up in Linux 5.18 (and the stable kernels):
https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/commit/?id=30ffd5332e06 

Many thanks,
Oliver

closed 2022-03-17 by hartkopp