linux-can/can-utils #352: cansinffer 29bit?

David-Afik, 2022-03-15:
Is there any way to use cansinffer on 29bits IDs?
I saw there was a try of doing it - but I coudn't understand if it's working or not (right now on my Pi - it doesn't sniff the 29 bits , everything else is working )
when I try to enter the link - I get 404 error

 Thanks,

David-Afik, 2022-03-20:
I'm using PI 4 
with this kernel :

after reboot the PI I didn't get the `Protocol error` - so this is not a problem anymore 

I have looked at   [#250](https://github.com/linux-can/can-utils/issues/350)
but there is nothing there that explain what to do \ install 
I have the lateset updated on the pi (I think ):
sudo apt updte 
sudo apt full-upgrade 
sudo rpi-update

after reboot htis is what I have 

`Linux raspberrypi 5.15.28-v7l+ #1532 SMP Tue Mar 15 13:16:21 GMT 2022 armv7l GNU/Linux`
 
after purge the old can-utils and install it again I still have this version 
`
The following NEW packages will be installed:
  can-utils
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/103 kB of archives.
After this operation, 430 kB of additional disk space will be used.
Selecting previously unselected package can-utils.
(Reading database ... 104468 files and directories currently installed.)
Preparing to unpack .../can-utils_2018.02.0-1_armhf.deb ...
Unpacking can-utils (2018.02.0-1) ...
Setting up can-utils (2018.02.0-1) ...
Processing triggers for man-db (2.8.5-2) ...
`

so what am I missing \ need to do now ?

Thanks ,

hartkopp, 2022-03-20:
That you do not get the protocol error anymore is good!

The can-utils have not been upgraded in Debian for a while.

Therefore purge the can-utils again and build/install the can-utils from source.

- clone this repo (`git clone https://github.com/linux-can/can-utils.git`)
- `cd can-utils`
- `make`
- `sudo make install`

This installs the tools in `/usr/local/bin` which just works fine then.

David-Afik, 2022-03-20:
Thanks
it's seem to be working now!

is there any way to update Debian so it will install it? ot it's to much work? 

Thank you very much again!

hartkopp, 2022-03-20:
> Thanks it's seem to be working now!

Great! Have fun ;-)

> is there any way to update Debian so it will install it? ot it's to much work?

Alexander GQ Gerasiov @gerasiov is maintaining the Debian package: 
https://packages.debian.org/sid/net/can-utils

Maybe he sees this discussion and creates a new release (which also contains relevant updates for the upcoming Linux 5.18) ...

David-Afik, 2022-03-20:
I think it will be very helpful 
thank you!

closed 2022-03-23 by hartkopp