linux-can/can-utils #353: Incorrect clock value

lewisbennett, 2022-04-08:
I'm trying to tap into one of the CAN buses on a BMW F30 and a MINI R56 for use with an OpenAutoPro project, but haven't been able to get it to work so far. I'm using [this CAN HAT](https://thepihut.com/products/rs485-can-hat-for-raspberry-pi) attached to a Raspberry Pi 4, and for testing I've just been using vanilla Raspbian and not the OpenAutoPro distributions.

What I'm currently struggling with is that using `candump can0` doesn't show any messages, so I'm troubleshooting to see where things might be going wrong. One of the things I noticed is that the clock value is wrong.

Inside `/boot/config.txt` I have the following lines:
```
dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=12000000,interrupt=25
```

However, the output of `ip -d link show can0` shows a clock value of `6000000` - exactly half what I enter in the boot config:
```
4: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0
    can state STOPPED restart-ms 0
          mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
          clock 6000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
```

Interestingly, running `cat /sys/kernel/debug/clk/clk_summary` shows `can0_osc` with the same value as the boot config.

The above is repeatable; any value I enter in the boot config is divided by two in the output of the above command. Just for fun, I doubled the value to `24000000` (so that it would show the *correct* value) and, after running `ip link set can0 up type can bitrate 500000`, I was able to receive packets from a DIY CAN node I built using an Arduino (this wasn't previously possible, but still doesn't work with the car though).

I'm not sure whether this is the root cause of the issues I'm having however, I've not managed to find any other mention of this online, yet it's the only way I've been able to get any sign of life out of it, despite all online instructions saying to use the exact value of the oscillator.

Cheers!

marckleinebudde, 2022-04-08:
Hey @lewisbennett,

according to the user manual, there are 2 versions of the board, one with a 8 MHz and one with a 12 MHz oscillator. Please ensure that you have the 12 MHz variant.

In the output of `ip -d link show can0` you see the effective clock rate that's used to generate the bit timing. The mcp2515 has an internal divider of 2, which results in the effective clock rate being the half of the external one.

lewisbennett, 2022-04-08:
Thanks for the quick response!

It's definitely the 12MHz variant. I wasn't aware of the internal divider, so that explains the output. I take it this is completely normal/expected behaviour then?

marckleinebudde, 2022-04-08:
At least the `clock 6000000` for a 12 MHz oscillator is expected, a not working device with your car is not :smile:.

I suggest to build a CAN bus with your Arduino first, using the same bitrate as in you car's bus. If this is working attach the Raspi and your Arduino to the bus you want to tap into. If there CAN frames on that bus, you should see them on the Arduino and your Raspi.

lewisbennett, 2022-04-08:
Unfortunately I've already tried solely using the Arduino to read frames from the car with no success. Guess the issue lies somewhere else 🤷‍♂️

I'll close this issue now. Thanks for clarifying so quickly!

closed 2022-04-08 by lewisbennett