linux-can/can-utils #356: Writing a CAN Frame with an invalid CAN ID to a raw CAN socket does not raise on error

derek-will, 2022-04-16:
An improvement that I think could be made is when a caller configures a CAN ID in a CAN Frame to be invalid that SocketCAN returns -1 in a subsequent write attempt and sets an appropriate errno such as EINVAL.

For example, configuring a CAN ID to exceed 11-bit without setting the CAN_EFF_FLAG should be caught and thrown back to the caller as an error. Currently if a caller defines a CAN Frame with CAN ID > 0x7FF and does not set the CAN_EFF_FLAG, then SocketCAN will consume the CAN Frame in a `write` \ `send` function call and silently truncate the CAN ID to 11-bit.

I think this change will help application programmers by flagging this "programmer error" and not adjusting the input to something they likely didn't mean to send.

Please let me know what you think of this suggestion. If accepted as concept, I volunteer to make and test the change.

olerem, 2022-04-16:
Are you talking about CAN_RAW socket? The CAN_RAW socket driver do not drops CAN ID. It takes prepared RAW frame and send it. On other hand, cansend will parse the input silently drops some bits:
https://github.com/linux-can/can-utils/blob/master/lib.c#L187
On this place you can probably add some validation.

derek-will, 2022-04-18:
What I am referring to is something like the following:

```
frame.can_id  = 0x1500; // Don't set CAN_EFF_FLAG
frame.can_dlc = 2;
frame.data[0] = 0x11;
frame.data[1] = 0x22;

nbytes = write(s, &frame, sizeof(struct can_frame));
```
Then the following will be transmitted onto the network.

> 500   [2]  11 22

This should be caught as an error and prevented.

Adding some logic to `raw_sendmsg` may be able to accomplish this: https://elixir.bootlin.com/linux/latest/source/net/can/raw.c#L771

hartkopp, 2022-04-18:
There are two ways you might create and send 'handcrafted' CAN(FD) frames:

1. PF_CAN / CAN_RAW socket
2. PF_PACKET socket

When you are using the virtual CAN (e.g. vcan0) you just get the stuff back that you sent (garbage in - garbage out) which is intentional to test the CAN network layer. E.g. you can create error message frames and push them through the stack for testing.

At the point where the CAN frame goes through a 'real' CAN device (RX/TX) the values are automatically sanitized by the CAN controller's registers. So the 'real' CAN traffic is always ok.

The question is what happens to existing applications when we now enable such 'enforcing' by default - and what to check then?

derek-will, 2022-04-21:
Hi @hartkopp,

I hooked up my USB2CAN device and an engine controller on my bench.

I implemented the following:

```
frame.can_id  = 0x1500; // Don't set CAN_EFF_FLAG
frame.can_dlc = 8;
frame.data[0] = 0x01;
frame.data[1] = 0x23;
frame.data[2] = 0x45;
frame.data[3] = 0x67;
frame.data[4] = 0x89;
frame.data[5] = 0xAB;
frame.data[6] = 0xCD;
frame.data[7] = 0xEF;

nbytes = write(s, &frame, sizeof(struct can_frame));
```

nbytes is set to 16 after the call completes and I see the following in candump:

> candump can0,500:7FF
> can0  500   [8]  01 23 45 67 89 AB CD EF

I think consolidation of this sort of check in SocketCAN core would be ideal as it would cover all devices against this sort of invalid input due to "programmer error" regardless of the device driver.

hartkopp, 2022-04-21:
Hi @derek-will ,

I got your point even without the code example.

But I can not see a real gain when you are able to create this kind of 'broken' frames over PF_PACKET sockets anyway.

derek-will, 2022-05-01:
Hi @hartkopp, I did not know about the PF_PACKET sockets capability with SocketCAN. I apologize for my ignorance on this. Are there any examples of this capability available? I can't seem to find them in documentation or example code anywhere.

It will be no big deal to add the check in the wrapper library logic anyway. I appreciate your time and consideration on this topic. Thanks as always!

closed 2022-05-01 by derek-will

hartkopp, 2022-05-02:
Yes :-)

Here we are: https://github.com/linux-can/can-tests/blob/master/netlayer/tst-packet.c

And the discussion with you is really valuable. Thx!

derek-will, 2022-05-15:
Would you be opposed to adding something like this for ISO-TP sockets? 

I am assuming that PF_PACKET approach only works for Raw CAN messages. If that is the case, then there woud be value to preventing SockAddr structs which were incorrectly formatted from binding on ISO-TP sockets in the first place. This came to me as I was authoring the verification code in my .NET wrapper project this morning. Let me know your thoughts either way. If accepted, I would volunteer to do the legwork of patching, testing, etc. (and signing off this time :-P)

hartkopp, 2022-05-15:
The isotp implementation does CAN-ID sanitizing now: https://github.com/torvalds/linux/commit/3ea566422cbde9610c2734980d1286ab681bb40e

So when the application programmer feeds any trash into the CAN ID fields it does not harm the system anymore ;-)

derek-will, 2022-05-15:
Ah! Lol. That one is funny. :-D

What I am referring to is this:

If I set my CAN ID pair to 0x17E0 / 0x7E8 on my ISO-TP socket, then anything after the 11th bit is masked off when the CAN_EFF_FLAG is not set. 

Example output in candump:

```
  vcan0  7E0   [2]  01 03
  vcan0  7E8   [8]  10 0E 43 06 01 43 01 96
  vcan0  7E0   [3]  30 00 00
  vcan0  7E8   [8]  21 02 34 02 CD 03 57 0A
  vcan0  7E8   [2]  22 24
```

This problem was actually an enabler of that previous one. When the application programmer sets anything beyond the 11th bit that isn't an RTR, EFF, or ERR flag, then it ought to be bounced back to them as an error when the EFF flag is not set. In the most likely case they just forgot to set the EFF flag. I know that early on in my use of SocketCAN, I myself was bit by this a few times until I learned better. 

How I am handling this in the Land O' .NET is:
```
public static void ThrowIfCanIdStructureInvalid(uint canId)
{
    uint rawId = canId & SocketCanConstants.CAN_EFF_MASK; // remove ERR, RTR, and EFF flags
    if (rawId > 0x7ff && (canId & (uint)CanIdFlags.CAN_EFF_FLAG) == 0)
    {
        throw new ArgumentException("CAN_EFF_FLAG must be set when CAN ID exceeds 11 bits.", nameof(canId));
    }
}
```
So although the ISO-TP implementation is now sanitizing the input, it is still not validating that it is plausible. Does that make sense?

hartkopp, 2022-05-15:
So you are thinking about such an extension?

```
diff --git a/net/can/isotp.c b/net/can/isotp.c
index 35a1ae61744c..f0029229289f 100644
--- a/net/can/isotp.c
+++ b/net/can/isotp.c
@@ -1180,10 +1180,15 @@ static int isotp_bind(struct socket *sock, struct sockaddr *uaddr, int len)
        if (rx_id & CAN_EFF_FLAG)
                rx_id &= (CAN_EFF_FLAG | CAN_EFF_MASK);
        else
                rx_id &= CAN_SFF_MASK;
 
+       /* give feedback on wrong CAN-ID values */
+       if (tx_id != addr->can_addr.tp.tx_id ||
+           rx_id != addr->can_addr.tp.rx_id)
+               return -EINVAL;
+
        if (!addr->can_ifindex)
                return -ENODEV;
 
        lock_sock(sk);
 
```

derek-will, 2022-05-15:
That code is amazingly effective! It would both solve this issue (implausible CAN IDs) as well as notify the caller in the event that they erroneously set the RTR and ERR flags as those are not supported.

hartkopp, 2022-05-15:
:-D Thanks!
Will send a patch for the can-next tree, which will not be applied to stable trees as it is an extension - and not something that really fixes something.

Will add a "Suggested-by:" tag for you then, ok?

derek-will, 2022-05-15:
Sounds good and thanks again! Let me know if you want me test or anything.

hartkopp, 2022-05-15:
Patch is out: https://lore.kernel.org/linux-can/20220515181633.76671-1-socketcan@hartkopp.net/T/#u

The patch looks so easy but I'll boot in a minute to test it  ;-)